OpenDroneMap/ODM #589: GCP text file - software looks in the 'code' folder?

everydayduffy, 2017-05-24:
Hello,

I'm trying to run Opendronemap with a gcp text file as described in the wiki. 

I've launched the following command in Docker on Windows 10:

`docker run -it --rm -v C:/Software/OpenDroneMap/images:/code/images -v C:/Software/OpenDroneMap/odm_orthophoto:/code/odm_orthophoto -v C:/Software/OpenDroneMap/odm_texturing:/code/odm_texturing my_odm_image --min-num-features 12000 --use-pmvs --gcp C:/Software/OpenDroneMap/gcp_list.txt`

I got the following output:

Whole texturing procedure took: 156.798s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   C:/Software/OpenDroneMap/gcp_list.txt
[INFO]    Found /code/C:/Software/OpenDroneMap/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensf
imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inp
outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPoint
-outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_mode
erencing_log.txt -georefFileOutputPath /code/odm_georeferencing/odm_g
are/OpenDroneMap/gcp_list.txt -outputCoordFile /code/odm_georeferenci
Error in Georef:
Failed opening gcp file /code/C:/Software/OpenDroneMap/gcp_list.txt f

[ERROR]   Georeferencing failed.
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 146, in process
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Wed May 24 13:20:40  2017

Is this an issue with the formar of my gcp text file, or am I right in thinking that the software is looking in the 'code' folder by default instead of looking at the path I provided for the `--gcp` flag?

dakotabenjamin, 2017-05-26:
Docker and gcps are kinda weird. Try putting the gcp file in the images folder and then running the same command with `--gcp /code/images/gcp_list.txt`

dakotabenjamin, 2017-05-26:
Docker and gcps are kinda weird. You need to add the gcp file as a volume mount like this: ` -v C:/Software/OpenDroneMap/gcp_list.txt:/code/gcp_list.txt`. Since it is named in a standard fashion, you don't need to specify `--gcp`

everydayduffy, 2017-05-27:
Thanks for the help @dakotabenjamin. I ran the following:

`docker run -it --rm -v C:/Software/OpenDroneMap/gcp_list.txt:/code/gcp_list.txt -v C:/Software/OpenDroneMap/images:/code/images -v C:/Software/OpenDroneMap/odm_orthophoto:/code/odm_orthophoto -v C:/Software/OpenDroneMap/odm_texturing:/code/odm_texturing my_odm_image --min-num-features 12000 --use-pmvs --dem`

And it got as far as:

Generating texture patches:
        Running... done. (Took 4.661s)
        1111 texture patches.
Running global seam leveling:
        Create matrices for optimization...  done.
        Lhs dimensionality: 113230 x 113230
        Calculating adjustments:
                Color channel 1: CG took 100 iterations. Residual
                Color channel 0: CG took 100 iterations. Residual
                Color channel 2: CG took 101 iterations. Residual
                Took 0.216 seconds
        Adjusting texture patches 100%... done. (Took 1.05s)
Running local seam leveling:
        Blending texture patches 100%... done. (Took 17.938s)
Generating texture atlases:
        Sorting texture patches... done.
        Working on atlas 10 100%... done.
        Finalizing texture atlases... done. (Took: 4.09s)
Building objmodel:
        Saving model... done.
Whole texturing procedure took: 126.963s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /code
[INFO]    Found /code/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/op
imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400
outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputP
-outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_
erencing_log.txt -georefFileOutputPath /code/odm_georeferencing/o
.txt -outputCoordFile /code/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
[ERROR]   Georeferencing failed.
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 146, in proces
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Sat May 27 10:17:52  2017

Is this indicative of an issue with my `gcp_list.txt` file? Or something else?

everydayduffy, 2017-05-27:
My GCP file looks like this...
[gcp_list.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1033490/gcp_list.txt)

yjmenezes, 2017-05-27:
@everydayduffy 

you have 4 control points.

I plot and saw some are aligned. ( A, B and C )

try a short gcp_list.txt and check if point D is selected in best triplet or you will have altimetric problems. 

This is an issue i am observing with current georeferencing procedure.
[gcp_positions.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1033587/gcp_positions.zip)

yjmenezes, 2017-05-27:
@everydayduffy 

I translate your GCP to UTM with PROJ4.

[gcp_positions.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1033728/gcp_positions.zip)

everydayduffy, 2017-05-27:
@yjmenezes I got the same error using your modified version of the text file (WGS84 version). How can I re-run the georeferencing stage in docker without running the whole process each time? (it takes about 4 hours on my laptop). Then I can try your UTM version...

yjmenezes, 2017-05-27:
@everydayduffy 

you can try this:
a) remove both directories: 
rm odm_georeferencing   odm_orthophoto

b) use the gcp_list.txt with UTM

c) add this switchy:   --start-with odm_georeferencing 

BTW:  I have reported about PixelX_ and PixelY_ need to be measured or scaled to fit resized image.

everydayduffy, 2017-05-30:
@yjmenezes I couldn't get the `--start-with-odm_georeferencing` switch to work with docker. I added it to the end of my input command and it just completely ignored it and started extracting ccd info etc. etc.

Is a different syntax required?

yjmenezes, 2017-05-30:
@everydayduffy 

check your typed line.
 --start-with    StageToStart
 --start-with        odm_georeferencing

everydayduffy, 2017-05-31:
@yjmenezes - That doesn't work in Docker. e.g.

`docker run -it --rm -v C:/Software/OpenDroneMap/images:/code/images -v C:/Software/OpenDroneMap/odm_orthophoto:/code/odm_orthophoto -v C:/Software/OpenDroneMap/odm_texturing:/code/odm_texturing my_odm_image --opensfm-processes 4 --dem --min-num-features 12000 --start-with odm_orthophoto`

runs the whole process again...

yjmenezes, 2017-05-31:
@everydayduffy 

--start-with   works whell for me, i am using Ubuntu 16.04 guest under  QEMU .
the only issue:  I have to clean or remove the stage directory and sequence stages too.

for example:  --star-with  odm_georeference 
I have to clean both: odm_georeferencing/*  and odm_orthophoto/*
All previous stage warning me about a valid file found.

Under Docker, the --rm clean the Container. Try run without   --rm.

btw:
--orthophoto-resolution  10.0 \
--dem --dem-sample-radius 0.25 --dem-resolution 0.2 --dem-radius 1.0 
works for me too.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/gcp-text-file-software-looks-in-the-code-folder/395

closed 2018-03-28 by dakotabenjamin