OpenDroneMap/ODM #590: Error processing images with docker container

gediminasgu, 2017-05-26:
Hi,

I use official docker container to process my images with following command:
``docker run -it --rm -v $(pwd)/geotagged:/code/images -v $(pwd)/odm_ortophoto:/code/odm_orthophoto opendronemap/opendronemap``

"geotagged" folder contains my images geotagged with MissionPlanner. "odm_ortophoto" is empty folder. After running that command line I get following error:
```
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /code/odm_texturing/odm_textured_model.obj
[DEBUG]   running /code/SuperBuild/install/bin/texrecon /code/opensfm/reconstruction.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/code/SuperBuild/install/bin/texrecon (built on May  9 2017, 18:15:20)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 23946 verts... 47715 faces... done.
Warning: Zero-length normals detected: 88 face normals, 0 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 9
NVM: Number of features: 0
        Loading 0%...terminate called after throwing an instance of 'util::Exception'
terminate called recursively
Aborted (core dumped)
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/mvstex.py", line 109, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 134
```
I have tried two different datasets and I get the same error with both. Maybe the problem could be, because I use Raspberry Pi to take the photos, which doesn't include lens information into a exif?

Any help would be appreciated.

GG

yjmenezes, 2017-05-29:
@gediminasgu 

i have processed scanned images without exif tags.
Check, with exiftool,  if your tagged images has Focal Lenght or if your camera name and model are into  .../opensfm/sensor_data.json and  .../opensfm/sensor_data_detailed.json

a) you could also try those switches:
--force-focal  FOCAL_mm    --force-ccd   CCD_WIDTH_mm
or
b) write exif tags into your files, for this, edit my script.
https://github.com/yjmenezes/odm_tnova/blob/master/shell/lmk_exif_from_lonlat.sh

i hope it helps

dakotabenjamin, 2017-06-07:
There could be issues stemming from the raspberry pi camera. can you run: 
```
docker run -it --rm -v $(pwd)/geotagged:/code/images -v $(pwd)/opensfm:/code/opensfm opendronemap/opendronemap
```
and then load the opensfm/depthmaps/merged.ply into MeshLab (or other point cloud viewing software?

gediminasgu, 2017-06-08:
Thanks, I'll try it when I'll back home.
BTW, I have tried to pass parameters ``--force-focal FOCAL_mm --force-ccd
CCD_WIDTH_mm`` but it didn't help, I got the same error.

yjmenezes, 2017-06-08:
@gediminasgu 

your exif tags shows  focal= 3mm AND Image Size: 3280x2464.

please confirm ccd_width, i found this
The active pixels of the sensor chip are 2592 x 1.4um = 3.629mm wide. 

[exifinfo.txt.gz](https://github.com/OpenDroneMap/OpenDroneMap/files/1061226/exifinfo.txt.gz)
ccd_width: 3280 x 1.4 = 4.592 mm

pierotofy, 2020-09-11:
ODM can handle non-georeferenced images quite well (now featuring pseudo geofererencing for non-georeferenced orthophotos). If there are still problems, please re-open :pray:

closed 2020-09-11 by pierotofy