OpenDroneMap/ODM #591: [Windows] docker: invalid reference format

AZOM, 2017-05-29:
After having problems running ODM with the vagrant variant, I thought I'd give docker a try. Unfortunately I have following problem which I cannot solve:

`foo@bar MINGW64 /c/workspace/bellus
$ echo $(pwd)
/c/workspace/bellus

foo@bar MINGW64 /c/workspace/bellus
$ ls -al
total 6
drwxr-xr-x 1 azollike 1049089   0 May 29 14:13 ./
drwxr-xr-x 1 azollike 1049089   0 May 29 14:21 ../
-rw-r--r-- 1 azollike 1049089  17 May  7  2016 .gitignore
-rw-r--r-- 1 azollike 1049089 321 May  7  2016 gcp_list.txt
drwxr-xr-x 1 azollike 1049089   0 May 29 14:13 images/

foo@bar MINGW64 /c/workspace/bellus
$ ls images/
IMG_1297_RGB.jpg  IMG_1310_RGB.jpg  IMG_1323_RGB.jpg  IMG_1336_RGB.jpg  IMG_1349_RGB.jpg  IMG_1362_RGB.jpg  IMG_1375_RGB.jpg  IMG_1388_RGB.jpg  IMG_1401_RGB.jpg  IMG_1414_RGB.jpg
IMG_1298_RGB.jpg  IMG_1311_RGB.jpg  IMG_1324_RGB.jpg  IMG_1337_RGB.jpg  IMG_1350_RGB.jpg  IMG_1363_RGB.jpg  IMG_1376_RGB.jpg  IMG_1389_RGB.jpg  IMG_1402_RGB.jpg  IMG_1415_RGB.jpg
IMG_1299_RGB.jpg  IMG_1312_RGB.jpg  IMG_1325_RGB.jpg  IMG_1338_RGB.jpg  IMG_1351_RGB.jpg  IMG_1364_RGB.jpg  IMG_1377_RGB.jpg  IMG_1390_RGB.jpg  IMG_1403_RGB.jpg  IMG_1416_RGB.jpg
IMG_1300_RGB.jpg  IMG_1313_RGB.jpg  IMG_1326_RGB.jpg  IMG_1339_RGB.jpg  IMG_1352_RGB.jpg  IMG_1365_RGB.jpg  IMG_1378_RGB.jpg  IMG_1391_RGB.jpg  IMG_1404_RGB.jpg  IMG_1417_RGB.jpg
IMG_1301_RGB.jpg  IMG_1314_RGB.jpg  IMG_1327_RGB.jpg  IMG_1340_RGB.jpg  IMG_1353_RGB.jpg  IMG_1366_RGB.jpg  IMG_1379_RGB.jpg  IMG_1392_RGB.jpg  IMG_1405_RGB.jpg  IMG_1418_RGB.jpg
IMG_1302_RGB.jpg  IMG_1315_RGB.jpg  IMG_1328_RGB.jpg  IMG_1341_RGB.jpg  IMG_1354_RGB.jpg  IMG_1367_RGB.jpg  IMG_1380_RGB.jpg  IMG_1393_RGB.jpg  IMG_1406_RGB.jpg
IMG_1303_RGB.jpg  IMG_1316_RGB.jpg  IMG_1329_RGB.jpg  IMG_1342_RGB.jpg  IMG_1355_RGB.jpg  IMG_1368_RGB.jpg  IMG_1381_RGB.jpg  IMG_1394_RGB.jpg  IMG_1407_RGB.jpg
IMG_1304_RGB.jpg  IMG_1317_RGB.jpg  IMG_1330_RGB.jpg  IMG_1343_RGB.jpg  IMG_1356_RGB.jpg  IMG_1369_RGB.jpg  IMG_1382_RGB.jpg  IMG_1395_RGB.jpg  IMG_1408_RGB.jpg
IMG_1305_RGB.jpg  IMG_1318_RGB.jpg  IMG_1331_RGB.jpg  IMG_1344_RGB.jpg  IMG_1357_RGB.jpg  IMG_1370_RGB.jpg  IMG_1383_RGB.jpg  IMG_1396_RGB.jpg  IMG_1409_RGB.jpg
IMG_1306_RGB.jpg  IMG_1319_RGB.jpg  IMG_1332_RGB.jpg  IMG_1345_RGB.jpg  IMG_1358_RGB.jpg  IMG_1371_RGB.jpg  IMG_1384_RGB.jpg  IMG_1397_RGB.jpg  IMG_1410_RGB.jpg
IMG_1307_RGB.jpg  IMG_1320_RGB.jpg  IMG_1333_RGB.jpg  IMG_1346_RGB.jpg  IMG_1359_RGB.jpg  IMG_1372_RGB.jpg  IMG_1385_RGB.jpg  IMG_1398_RGB.jpg  IMG_1411_RGB.jpg
IMG_1308_RGB.jpg  IMG_1321_RGB.jpg  IMG_1334_RGB.jpg  IMG_1347_RGB.jpg  IMG_1360_RGB.jpg  IMG_1373_RGB.jpg  IMG_1386_RGB.jpg  IMG_1399_RGB.jpg  IMG_1412_RGB.jpg
IMG_1309_RGB.jpg  IMG_1322_RGB.jpg  IMG_1335_RGB.jpg  IMG_1348_RGB.jpg  IMG_1361_RGB.jpg  IMG_1374_RGB.jpg  IMG_1387_RGB.jpg  IMG_1400_RGB.jpg  IMG_1413_RGB.jpg

foo@bar MINGW64 /c/workspace/bellus
$ winpty docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --meshing-size 100000
docker: invalid reference format.
See 'docker run --help'.
`

Can anyone help please?

patrickelectric, 2017-05-30:
In windows is \ and not / to folders.
Eg. `docker run -it --rm -v C:\Users\Usuario\git\odm_data_seneca\images:/code/images opendronemap/opendronemap`

AZOM, 2017-06-06:
@patrickelectric Unfortunately that has still the same result:
`winpty docker run -it --rm -v C:\\workspace\\bellus\\images:/code/images -v C:\\workspace\\bellus\\odm_orthophoto:/code/odm_orthophoto -v C:\\workspace\\bellus\\odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --meshing-size 100000`

results in:

> docker: invalid reference format.
See 'docker run --help'.

And I have to use \\ double backspace because if you only do one it is recognized as escaping and the path is wrong. I tested with `ls` and that works without issues.

dakotabenjamin, 2017-06-07:
What is winpty? is that potentially affecting the docker command?

AZOM, 2017-06-19:
@dakotabenjamin This an additional command you need to use when operating under Windows to get a bash shell like response from whatever you are executing afterwards (e.g. docker). See https://github.com/rprichard/winpty

Icemap, 2017-09-12:
@AZOM Please tell me the final solution. THX.

tuxrace, 2017-09-13:
@Icemap @AZOM  try wrapping with double quotes and adding a forward slash ```"/$(pwd)"```
```winpty docker run -it --rm -v "/$(pwd)"/images:/code/images -v "/$(pwd)"/odm_orthophoto:/code/odm_orthophoto -v "/$(pwd)"/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --meshing-size 100000```

irozgar, 2017-09-15:
@Icemap I was having this issue and the suggestion made by @tuxrace didn't work, but I've managed to solve this by changing the slashes in the windows path. So instead of
`docker run -it --rm -v D:\Users\irozgar\Documents\Le
arningDocker\php-app:/opt -w /opt --network=phpapp_appnet shippingdocker/php ls` I used `docker run -it --rm -v D:/Users/irozgar/Documents/Le
arningDocker/php-app:/opt -w /opt --network=phpapp_appnet shippingdocker/php ls

KevinBurton, 2017-10-20:
I am still having this same problem with docker on window

> docker run -it -v C:/Users/rburton/code/Docker/R:/home/docker -w /home/docker -u docker --rm /usr/bin/bash

> ... docker.exe:  invalid reference format.

> Server Version: 17.09.0-ce

Arnotjevleesch, 2017-10-25:
I face the same problem with jenkins on powershell or git bash

- powershell
docker run --name jenkins -p 8080:8080 –v c:/Users/Arnaud/Desktop/docker_data:/var/jenkins_home jenkinsci/jenkins:2.73.2

- git bash
docker run --name jenkins -p 8080:8080 –v //c/Users/Arnaud/Desktop/docker_data jenkinsci/jenkins:2.73.2

pavloneiman, 2017-11-12:
It should work after adding forward slash in from of windows path
```
docker run -it -v /C:/Users/rburton/code/Docker/R:/home/docker -w /home/docker -u docker --rm /usr/bin/bash
```

alyssaong1, 2018-02-05:
Also note that if you have spaces in your folder name this error occurs

tapaz1, 2018-02-27:
@irozgar 's answer worked for me. I was using $(pwd):/app in my volumes declaration, which is the present working directory with Windows backslashes included. Explicitly typing the path fixed the issue.

e.g. -v C:\your\app\directory\:/app

Hope this helps others

closed 2018-03-16 by smathermather

BenMitchell1979, 2018-04-02:
Use ${pwd} vs $(pwd)  

`
docker run -v ${pwd}:/app/ -w /app/ -i -t hashicorp/terraform:light --version
`

mkborregaard, 2018-05-04:
I just stumbled across this thread and was super happy to find the solution.

gparvind, 2019-05-13:
putting the pwd variable and target path both in double quotes worked for me in powershell since I had whitespace in the source path.
`-v "${pwd}:/data"`

rom5jp, 2020-02-16:
I just put double quotes around the "host_path:container_path".
It worked for me in powershell.

j0ruge, 2020-03-31:
docker container run -p 8080:80 -v //D/Dev/volume:/usr/share/nginx/html nginx

anindyaCISPL, 2020-11-24:
> @AZOM
> @smathermather
> 
> If you want to reference a variable (such as calling your pwd) to mount the local fs.. In powershell you'll want to use ${pwd} vs $(pwd).. I'm putting this out there just in case anyone else stumbles across this thread.
> 
> `docker run -v ${pwd}:/app/ -w /app/ -i -t hashicorp/terraform:light --version`

It's worked
docker run -it --rm --name myphp -v ${pwd}:/usr/src/myapp -w /usr/src/myapp php:7.4-cli php index.php
(the reference given on docker hub : https://hub.docker.com/_/php

Thanks! @BenMitchell1979

mic7x4, 2021-01-11:
This worked for me after adding quotes and put all path toLowercase: 
docker container run -d -p 8080:80 -v "$(pwd):/user/share/nginx/html".ToLower() --name nginx-website nginx

FranciscoHernandez1998, 2021-02-17:
For me, while on Windows 10, it worked when I enclosed the full path with "", not adding ToLower().

i.e.
docker run -d -p 8080:80 -v "C:\Users\usr\Desktop\nginx container:/usr/share/nginx/html" --name nginx-website nginx

And having my own made **index.html** at the folder named **nginx container** in my desktop.

gareth-rogers-reclique, 2022-01-14:
This may or may not help windows users.  If you happen to have a path var (ex. %USERPROFILE%) and your the variable is separated by a space (ex. firstname lastname), put double-quotes around the path var.  This took me 30 mins to figure out :(

steelx, 2022-03-23:
WIndows Terminal
`docker run -v "$(pwd):\app" alpine cat "/app/hello.txt"`

DeepakSingh9, 2023-01-13:
I tried all the permutations and combinations on windows 11 . Still not running.
`$ docker run -it \
      -e POSTGRES_USER = "root" \
      -e POSTGRES_PASSWORD = "root" \
      -e POSTGRES_DB="ny_taxi" \
      -v "C:\Users\indee\Desktop\Data_Eng_Self\Deproject\ny_taxi_postgres_data:/var/lib/postgresql/data" \
      -p 5432:5432 \
       postgres:13
docker: invalid reference format.
See 'docker run --help'.`