OpenDroneMap/ODM #592: 'Killed' on orthophoto generation

everydayduffy, 2017-05-30:
Running ODM with docker on Windows 10. Ran the following:

```
docker run -it --rm -v C:/Software/OpenDroneMap/images:/code/images -v C:/Softwar
g my_odm_image --opensfm-processes 4 --dem --min-num-features 12000 --orthophoto-resolution 2500
```
(Have recently upgraded RAM so dropped the `--pmvs` switch)

The processing got so far and printed the following:

[DEBUG]   running /code/SuperBuild/build/pdal/bin/pdal pipeline -i /code/odm_georeferencing/las.json --readers.ply.file
ame=/code/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/code/odm_georeferencing/odm_georeferen
ed_model.las
[DEBUG]   running /code/SuperBuild/build/pdal/bin/pdal pipeline /code/odm_georeferencing/dem.json --readers.las.filenam
=/code/odm_georeferencing/odm_georeferenced_model.las --writers.gdal.filename=/code/odm_georeferencing/odm_georeferenci
g_model_dem.tif
[INFO]    DEM created at /code/odm_georeferencing/odm_georeferencing_model_dem.tif
[INFO]    Creating geo-referenced CSV file (XYZ format)
[WARNING] Found a valid georeferenced model in: /code/odm_georeferencing/odm_georeferenced_model.ply
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /code/odm_texturing/odm_textured_model_geo.obj -logFile /co
e/odm_orthophoto/odm_orthophoto_log.txt -outputFile /code/odm_orthophoto/odm_orthophoto.png -resolution 2500.0  -output
ornerFile /code/odm_orthophoto/odm_orthophoto_corners.txt
Killed
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_orthophoto.py", line 74, in process
    '-outputCornerFile {corners}'.format(**kwargs))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

My odm_orthophoto and odm_texturing folders were empty before running, so not sure why it killed itself. If anyone has advice, that would be much appreciated.

Thank you.

yjmenezes, 2017-05-30:
@everydayduffy 

you set  --orthophoto-resolution    2500.
2500 pixel by square meter is too much (2cmx2cm). 
ODM default is 20. ( pixel~ 22cmx22cm)
you could take your original image GSD as reference.

everydayduffy, 2017-05-31:
@yjmenezes - My original GSD is 0.79cm/pixel. 

So that's why I wanted something high res like 2500!

yjmenezes, 2017-05-31:
@everydayduffy 

your output file will be very big using 2500/square meter.
you could try test all process starting from 25, 250, than 500, and so on.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/killed-on-orthophoto-generation/391

closed 2018-03-28 by dakotabenjamin