OpenDroneMap/ODM #593: Error in OdmMeshing: Argument '-inputFile' has a bad value.

bryanibit, 2017-06-01:
When I ran odm_slam module, the error happened. Such as the following:
[INFO]    Running ODM PMVS Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /home/inin/OpenDroneMap/tests/video/odm_meshing/odm_mesh.ply
[DEBUG]   running /home/inin/OpenDroneMap/build/bin/odm_meshing -inputFile /home/inin/OpenDroneMap/tests/video/opensfm/depthmaps/merged.ply -outputFile /home/inin/OpenDroneMap/tests/video/odm_meshing/odm_mesh.ply -logFile /home/inin/OpenDroneMap/tests/video/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9  -samplesPerNode 1.0 -solverDivide 9
Error in OdmMeshing:
Argument '-inputFile' has a bad value. (file not accessible)
For more detailed information, see log file.
Traceback (most recent call last):
  File "run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/inin/OpenDroneMap/scripts/odm_meshing.py", line 74, in process
    '-samplesPerNode {samples} -solverDivide {solver}'.format(**kwargs))
  File "/home/inin/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

I am sure that I do not have spaces in file name, and I have install ODM correctly, so this is not similar to #64. Wish some one could help me.

dakotabenjamin, 2017-06-07:
What is the output of `ls -al /home/inin/OpenDroneMap/tests/video/opensfm/depthmaps/merged.ply`

thaytan, 2017-08-17:
I'm running into the same problem. It's my first time looking at OpenDroneMap, but I think what's happening for me is that I'm running from video, and opensfm cell is never getting run, so the opensfm/depthmaps/merged.ply file isn't created. It certainly doesn't exist for me.

alke, 2017-12-23:
there is no /opensfm/depthmaps/merged.ply it is only /pmvs/recon0/models/option-0000.ply
if I use "--use-pmvs"  "python run.py --project-path /home/xxxx/Phantom_Dji/3d/projekt --video DJI00015.MP4 --slam-config config.yaml --resize-to 1920 --use-pmvs testv" it runss thru all

pierotofy, 2020-09-11:
These issues should have been resolved (the odm_meshing app no longer exists!). :smile: