OpenDroneMap/ODM #594: tif support + mesh smoothing

tigallouche, 2017-06-02:
Hi,
I used 16bit tif photos and ran into issues when running ODM. I have made changes to fix these issues (it was mostly related to exif extraction not working well). Now it runs without trouble.

I also added an extra step to the meshing process that uses pcl laplacian smoothing function to smooth the mesh. I found it dramatically improves the quality of orthophotos for farm land surveys. 

I have the commits on my local branch of the project and I'd like to share my work if ppl are interested.

My apologies if there is an obvious process to do so - I am quite new to github and to the ODM community.

yjmenezes, 2017-06-02:
@tigallouche 

1- I have faced exif problems too. Could you please explain how did you solve it ?
I have  processed without exif and used GCP later.

2- have you tested if laplacian filter improve contours apparence, cutting off some   very small islands and reduccing roughness ?

i will look your branch. thx.

tigallouche, 2017-06-03:
1. for exif :
 - added tif to supported formats in context.py
 - set option 'comment=false' when copying metadata in resize.py (since tif does not support image comments) : 
`old_meta.copy(comment=False)`
 - switched from 'jhead -v' to 'exiftool -n' in utmExtractor.cpp (jhead only supports jpeg?) - that fixes errors in georeferencing (I also had to write a new function to extract GPS lat, lon, alt, but since exiftool provides outputs with decimals it turned out to simplify the extraction).

2. I am not sure I fully understand the effects you are describing. Essentially, I added the smoothing because I mostly use ODM to build farm land orthophotos, and I noticed the smoother (flatter) the mesh, the cleaner the projected orthophoto (textures and not stretched nor blurred, no weird bubbles or other numerical artifacts, ...)

yjmenezes, 2017-06-03:
@tigallouche 

thank you. i will try your exif directions.

Please, take a look on the effects i am describing.
[image]

water bodies are a big problem too, besides trees and buildings.

dakotabenjamin, 2017-06-07:
@tigallouche You will want to fork the project to your own github account, push the local changes to that, then make a pull request. I'll test and merge, and if there are problems I'll try to work you through them.

PeterSprague, 2017-07-31:
Has this gotten pulled yet?  I didn't see it on the list.

I also would like this ability, and could duplicate the above work if this issue resolution is stalled.

dakotabenjamin, 2017-07-31:
Has not been pulled yet. @tigallouche can you do that? Very interested in your changes. Even if its not complete, we can collaborate to make it merge-worthy

pierotofy, 2020-02-03:
TIFF file support is now implemented!

For smoothing, reduce the `--mesh-octree-depth` option.

Please open a new issue if there are issues with the mesh still.

closed 2020-02-03 by pierotofy