OpenDroneMap/ODM #597: georeferenced mtl file sets diffuse and ambient to black

fredlllll, 2017-06-09:
the mtl file for the georeferenced model sets the diffuse and ambient color to black, while the normal mtl has it set to white.

as the mtl files dont differ otherwise, why isnt it just using the normal one (mtllib odm_textured_model.mtl in obj) or copies the normal one?

dakotabenjamin, 2017-06-09:
I haven't been able to find the source yet, but Ithink it will be in here somewhere: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_georef/src/Georef.cpp#L1099

fredlllll, 2017-06-09:
looked through the code, it doesnt seem to modify the values, but maybe it fails reading them correctly here? https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_orthophoto/src/OdmOrthoPhoto.cpp#L1206

yjmenezes, 2017-06-10:
[Georef301cpp.patch.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1065861/Georef301cpp.patch.zip)
@dakotabenjamin 

Georef.cpp has some issues.
This patch, over v3.0.1,  is just an output format enhancement and not a bug fix.
The line log_ <<"a)cam.focal_length shows were the gcp_list.txt  measured over original raw images will fail.
To fix it is necessary to replace cam.width by original raw image width.

closed 2020-04-14 by smathermather