OpenDroneMap/ODM #598: Comparison: ODM vs. DroneDeploy

AnthonyCoughlin, 2017-06-10:
Hey @pierotofy 

Wanted to give you guys some prospective on ODM vs. DroneDeploy.  First off, I updated WebODM this morning.  I captured these images this afternoon (10Jun17).  Its a small field house at a sports complex.  I processed the images through WebODM using the default settings and also processed them through DroneDeploy.  See the attached screenshots.  Also below, I've uploaded the images to Google Drive and provided a link if you'd like to inspect. 

Hope this helps further ODM's progress!
Anthony

[image]
[image]
[image]
[image]

https://drive.google.com/open?id=0B4Tf0aIvxRLJai1VM1VSUWV6Rm8

Fi156, 2017-06-11:
Photoscan:
No manual tie points were set. 

[image]
[image]

I really really dislike the drone deploy result: They are completely omitting roof overhang!
ODM however will produce much better results with parameter tweaking I guess.

AnthonyCoughlin, 2017-06-11:
Man, the Photoscan results look great.  What do you mean by "no manual tie-in points were set" (ground control points?) and how do I correct it!!??  I'm still learning the inner workings of ODM!

flyandi, 2017-07-21:
So what parameter tweaking do you have to do to achieve similar results? I am still trying to figure out the best angles to capture proper image data to generate high resolution 3d models from objects.

AnthonyCoughlin, 2017-07-21:
@Fi156, I would like to know more about "better parameter tweaking" as well.  I use force focal length for sure.  I am still having difficulty understanding and obtaining the correct CCD width for my on board camera.  If somebody could please advise.   For GPS I just utilize the EXIF data.  I also adjust the octree level depending on the level of detail.

Thanks in advance!

flyandi, 2017-07-21:
Whats your camera?

AnthonyCoughlin, 2017-07-21:
I fly Autel Robotics.  The EXIF data has the camera listed as XB004.  Some of my research has shown that the X-STAR uses the Sony Exmor-R sensor.  Correct me if I am wrong, but this sensor uses CMOS technology and not CCD and I'm not 100% if this effects the results of ODM.  However, on the Wikipedia page under the Exmor-R sensors section, there is a 12MP version so I use the dimension listed there (7.81mm)...

https://en.wikipedia.org/wiki/Exmor

https://www.bhphotovideo.com/c/product/1243948-REG/autel_robotics_xsprmor_x_star_premium_quadcopter_with.html

Thanks for the help.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/comparison-odm-vs-dronedeploy/377

closed 2018-03-28 by dakotabenjamin