OpenDroneMap/ODM #600: how to read dat from my computer

manguehdez, 2017-06-12:
Good morning I am a beginner in these so I wnat to learn about processing with ODM
I have al ready install ODM in Vagrant, but a couldn´t clone the sample data, so I download it but I dont know how to call it from ODM, another question is what kind of information do I have to need in the folder to run ODM, and make and orthomosaic, I am using a Phantom 4 to collect photos and PIX4D, so I need the photos, the log and waht else.

Thanks for everything.

yjmenezes, 2017-06-12:
@manguehdez 

you will need:
1- JPG images with EXIF tags to be placed inside  images/
2- camera informations: focal mm, sensor width and height in millimeters.
3- ground control points and its position measured over images. see gcp_list.txt

maybe you will have to use those switch at command line:  
--force-focal yourP4focalmm  --force-ccd yourP4ccdwictdmm    
and maybe
--use-exif   if you do not have  GroudControlPoints but your images have GPS exif tags.

take a look here https://github.com/OpenDroneMap/OpenDroneMap/wiki

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/how-to-read-dat-from-my-computer/388

closed 2018-03-28 by dakotabenjamin