OpenDroneMap/ODM #601: Coordinates file from EXIF data not working

geofis, 2017-06-12:
This is an issue related to OpenDroneMap 0.3.1. I run it from command line in Ubuntu 16.04; for installation, I followed instructions from the [main URL of the project](http://opendronemap.org/).

My goal is to generate an orthophoto from 4 geotagged images [(files here)](http://geografiafisica.org/consigna/visitante/opendronemap/tst.zip), for which I ran this code:

`python run.py tst1 --project-path /home/jr/ODMProjects/`

Everything went fine until the script attempted to generate coordinates file from EXIF data. The command line returned the following message:

> Generating coordinates file: /home/jr/ODMProjects/tst1/odm_georeferencing/coords.txt
> sh: 1: cannot create extract_utm_output.txt: Permission denied
> [WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
> Failed to open temporary jhead data file extract_utm_output.txt
> For more detailed information, see log file.
> [WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file. 
> [WARNING] Found a valid georeferenced model in: /home/jr/ODMProjects/tst1/odm_georeferencing/odm_georeferenced_model.ply
> 

Mesh and textured files were correctly generated, but couldn't find GTiff file in `odm_orthophoto`. I suppose this is related with permissions while executing the script, but couldn't find any suitable solution.

Any help would be appreciated.

dakotabenjamin, 2017-06-13:
Can you post the output for `ls -al /home/jr/ODMProjects/tst1/odm_georeferencing`?

dakotabenjamin, 2017-06-13:
I tried your images and it worked for me: 
[image]

So it may be a permissions issue on your end. BTW, I think this is the smallest number of images I've ever had successfully create an orthophoto. Usually I suggest no fewer than 9 images for a successful run.

yjmenezes, 2017-06-13:
@geofis 

try this:
python run.py tst1 --project-path /home/jr/ODMProjects/   --force-focal 3.6 --force-ccd 4.16  --use-exif

please, confirm --force-ccd  value, i am not sure it is 4.16
maybe --rerun-all too

geofis, 2017-06-13:
Thanks @dakotabenjamin @yjmenezes.

I tried these flags `--force-focal`, `--force-ccd` and `--use-exif`, but got the same result. Also tried `--rerun-all`, and again no luck.

Folder `odm_georeferencing` contains only one file:

> -rw-rw-r--  1 jr jr  407 jun 13 16:19 odm_georeferencing_utm_log.txt

With this contents:

> imagesPath was set to: /home/jr/ODMProjects/tst1/images_resize/
> imageListFile was set to: /home/jr/ODMProjects/tst1/opensfm/list_r000.out
> outputCoordFile was set to: /home/jr/ODMProjects/tst1/odm_georeferencing/coords.txt
> logFile_ was set to: /home/jr/ODMProjects/tst1/odm_georeferencing/odm_georeferencing_utm_log.txt
> Error in OdmExtractUtm:
> Failed to open temporary jhead data file extract_utm_output.txt

Certainly, it's a permission issue, as @dakotabenjamin on his test.

Maybe I should try installing ODM on a VM or a Docker, avoiding the typical "doesn't work on my machine".

I used few images only for testing, to get results quickly, but I know more is better. In fact, project has more than 200.

yjmenezes, 2017-06-13:
@geofis 

i run fine here under Ubuntu16.04
~/geofis       drwxr-xr-x
~/geofis/images   drwxr-xr-x

i run from ~/OpenDroneMap  ver 3.0.1 this way:
./run.sh  --project-path   ~/     geofis   --force-focal 3.6  --force-ccd 4.16 --use-exif

btw, do you know the P4 camera pixel size or ccd-width ?

dakotabenjamin, 2017-06-14:
Docker is great, I'd recommend using that.

geofis, 2017-06-14:
Thanks again @dakotabenjamin @yjmenezes.

Updating: definitely it's a permissions issue. Before I decided to run ODM in Docker, I gave it a try to @yjmenezes suggestion, and it worked!

Having images with +x...
> -rwxrwxr-x 1 jr jr 5067402 nov 26  2016 DJI_0004.JPG
> -rwxrwxr-x 1 jr jr 5606194 nov 26  2016 DJI_0005.JPG
> -rwxrwxr-x 1 jr jr 4908237 nov 26  2016 DJI_0006.JPG
> -rwxrwxr-x 1 jr jr 4993737 nov 26  2016 DJI_0007.JPG

...I ran this line:
`sudo ./run.sh tst1 --project-path /home/jr/ODMProjects/ --force-focal 3.6 --force-ccd 4.16 --use-exif`

Note the added `sudo` command, because without I received same errors from previous attempts. Now I have an ortophoto, virtually the same one posted by @dakotabenjamin:

[image]

This is the content of folder odm_georeferencing:

> -rw-r--r--  1 root root     153 jun 14 10:39 coords.txt
> -rw-r--r--  1 root root     296 jun 14 10:39 las.json
> -rw-r--r--  1 root root 5010419 jun 14 10:39 odm_georeferenced_model.csv
> -rw-r--r--  1 root root 4987807 jun 14 10:39 odm_georeferenced_model.las
> -rw-r--r--  1 root root 8126199 jun 14 10:39 odm_georeferenced_model.ply
> -rw-r--r--  1 root root    4097 jun 14 10:39 odm_georeferencing_log.txt
> -rw-r--r--  1 root root      29 jun 14 10:39 odm_georeferencing_model_geo.txt
> -rw-r--r--  1 root root     209 jun 14 10:39 odm_georeferencing_transform.txt
> -rw-r--r--  1 root root     319 jun 14 10:39 odm_georeferencing_utm_log.txt

I couldn't find the ccd width parameter from camera specification, so I used the one suggested by @yjmenezes. As this is only a test, using a wrong parameter is not a big concern for me now. Yesterday I searched for it, but only found that sensor is 1/2.3” CMOS from [specifications page](https://www.dji.com/phantom-4/info), This value is supposed to help me calculating ccd width, but couldn't find any suitable solution. Any suggestions for how to calculate it?

dakotabenjamin, 2017-06-14:
It's 6.17 mm

yjmenezes, 2017-06-14:
@geofis 

try compile ODM 0.3.1 as $ user
-rw-rw-r-- 1 jmenezes jmenezes 5067402 Nov 26  2016 DJI_0004.JPG

btw, i found this post about P4 camera:
http://forum.dji.com/thread-47292-1-1.html
fom post: """And NO I don't understand why DJI can't provide more exact specs. """

Focal length: 3.64mm
CMOS size: 6.25 x 4.68mm (7.81 diagonal)

could you, please run PMVS ?
 --use-pmvs   --pmvs-min-images 2

I run a QEMU under Debian8.6x64 host and Ubuntu16.04x64 guest  and ODM works very well.

geofis, 2017-06-17:
Thanks again.

Definitively, I'll compile as $ user in a virtual machine.

Also I tried with suggested pvms flags, and obtained a not desirably result, with patches located at wrong heights, and more blank areas.

[image]

Maybe this is because of the few images used, but there were no wrong patches in previous attempts without pvms flags.

yjmenezes, 2017-06-21:
@geofis 

could you please:
1- open ( I use QGIS )  odm_georeferenced_model.csv from opensfm and  pmvs.
2-  try add a second flight strip and  check model.csv again.
thx

geofis, 2017-06-22:
Finally I solved the permission issue, running ODM in a Xubuntu 16.04 VM. Everything seems to work fine now.

As suggested by @yjmenezes, I processed more images (37 overall), keeping pmvs flags, with this code:

`python run.py tst1 --project-path ~/ODMProjects --force-ccd 6.20 --use-exif --use-pmvs --pmvs-min-images 2`

"odm_georeferenced_model.csv" displayed in QGIS is as shown below:
[image]

And this was the result of the textured file, with floating patches again:
[image]

Using too few images is preventing a good result, but floating patches won't generate without pmvs flags.

yjmenezes, 2017-06-23:
@geofis 

good you solve permission issue.

could you please check openfsm "odm_georeferenced_model.csv"  ?

I have realised some gaps with opensfm vs pmvs ( my datasets ).
floating patch is another issue that needs to be tracked.

geofis, 2017-06-24:
@yjmenezes 

If I undertand the pipeline correctly, opensfm "odm_georeferenced_model.csv" is generated without pmvs flags, like in this code:
`run.py tst1 --project-path ~/ODMProjects --force-ccd 6.20`
...and this is how it looks like in QGIS:
[image]
(file size is ~62 MB)

And pmvs "odm_georeferenced_model.csv" is generated with this code:
`run.py tst1 --project-path ~/ODMProjects --force-ccd 6.20 --use-pmvs`
...which looks like this in QGIS:
[image]
(file size is ~17.5 MB)

pierotofy, 2019-06-21:
Closing this due to inactivity. The issues discussed here should be fixed. Try with an updated version of ODM, if the problem is still there please re-open a new issue.

closed 2019-06-21 by pierotofy