OpenDroneMap/ODM #602: Matching only the consecutive images in OpenSfM

AgarwalSaurav, 2017-06-13:
OpenSfM has an option matching_order_neighbors: n to match consecutive images only when the geoposition data is not available. See [this](https://github.com/mapillary/OpenSfM/issues/82). This is particularly helpful when frames are extracted from a video and it is known that the consecutive images are to be matched.

Is there a way we can activate this option while running OpenDroneMap? I believe the feature was introduced in OpenSfM on Jan 26. See [this](https://github.com/mapillary/OpenSfM/pull/133). I am not sure if ODM uses the latest OpenSfM.

dakotabenjamin, 2017-06-14:
We haven't updated OpenSfM to the newest changes yet, we are currently working with Mapillary on some updates and will have the newest changes added with that update.

AgarwalSaurav, 2017-06-14:
Ok thanks. Looking forward to it!

pierotofy, 2017-06-22:
Related, I think this would also solve https://github.com/OpenDroneMap/WebODM/issues/213

smathermather, 2020-04-14:
This is integrated.

closed 2020-04-14 by smathermather