OpenDroneMap/ODM #603: Problems isntalling

Gregoriomartjim, 2017-06-14:
Hello i am trying to install OpenDroneMap in a laptop with ubuntu 16.04
however i get this two errors while installing

[CMakeFiles/cgal.dir/all] Error 2
[cgal/stamp/cgal-download] Error 1

I been trying installing this libraries independently but it doesnt work 

could you help me out please

cheers

Gregorio Martinez

yjmenezes, 2017-06-15:
@Gregoriomartjim 

Qt5 is not used by ODM.
I compile today ODM 0.3.1
try:
1- upgrade your installed Ubuntu16.04, than  run  apt-get autoremove
2- remove OpenDroneMap directory
rm  -r  ~/OpenDroneMap
3- as $ user,  git clone ODV v0.3.1
4-  as $ user, run ./configure.sh  install

ODM 0.3.1 ./configure.sh install is running softly for me under Ubuntu . /etc/os-release 16.04.2  LTS

Gregoriomartjim, 2017-06-15:
I just reinstall ubuntu 16.04 and now it runs smoothly

Thanks

closed 2017-06-16 by dakotabenjamin