OpenDroneMap/ODM #604: errors running opendronemap

Gregoriomartjim, 2017-06-15:
I just install OpenDroneMap, but when i try to do the ./run.sh i got this error 

File "~/OpenDrone/run.py", line 8, in <module>
    import ecto
  File "~/OpenDrone/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: /home/rafaga/Escritorio/Gregorio/OpenDrone/SuperBuild/install/lib/libecto.so.0.6: undefined symbol: _ZN5boost7archive23basic_binary_iprimitiveINS0_15binary_iarchiveEcSt11char_traitsIcEE4loadERSs

could some one please help me out 

Regards

Gregoriomartjim, 2017-06-15:
I just reinstall ubuntu 16.04 and now it runs smoothly

dakotabenjamin, 2017-06-16:
Sounds good!

closed 2017-06-16 by dakotabenjamin