OpenDroneMap/ODM #607: Error: `Exception: Child returned 1`

kguidonimartins, 2017-06-18:
Dear,

I tried to run ODM following the guide to installation in my machine (`bash configure.sh install`). However, after this line:
`[DEBUG]   running PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/opensfm mesh /code/opensfm
`
the following error did appear: 

```
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/code/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 112, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
``` 
After this error, I tried to run using Docker (Option 2 of your wiki). Again, the same error above did appear.

Can you help me with this?

My settings:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
```

```
$ uname -r
4.4.0-79-generic
```

```
$ lshw -short
H/W path       Device  Class          Description
=================================================
                       system         Computer
/0                     bus            Motherboard
/0/0                   memory         7895MiB System memory
/0/1                   processor      Intel(R) Core(TM) i7-5500U CPU @ 2.40GHz
```

dakotabenjamin, 2017-06-19:
It looks like both installations worked, but there is an issue with your images. Can you tell me how many images you have, what overlap they were flown at, and also if possible can you share a sample image?

kguidonimartins, 2017-06-19:
I have 22 images with 85/75 frontal/side overlap. The images were taken at 60 meters high with a DJI Phantom 4 Pro. 

Image:
[image]

Image metadata: 
```
$ exiftool DJI_0001.JPG 
ExifTool Version Number         : 10.56
File Name                       : DJI_0001.JPG
Directory                       : .
File Size                       : 8.1 MB
File Modification Date/Time     : 2017:06:16 12:11:06-03:00
File Access Date/Time           : 2017:06:18 15:24:47-03:00
File Inode Change Date/Time     : 2017:06:16 17:29:24-03:00
File Permissions                : rw-r--r--
File Type                       : JPEG
File Type Extension             : jpg
MIME Type                       : image/jpeg
Exif Byte Order                 : Little-endian (Intel, II)
Image Description               : DCIM\100MEDIA\DJI_0001.JPG
Camera Model Name               : FC6310
Orientation                     : Horizontal (normal)
X Resolution                    : 72
Y Resolution                    : 72
Resolution Unit                 : inches
Software                        : v01.05.1577
Modify Date                     : 2017:06:16 15:11:07
Y Cb Cr Positioning             : Centered
Exposure Time                   : 1/240
F Number                        : 5.0
Exposure Program                : Program AE
ISO                             : 100
Exif Version                    : 0230
Date/Time Original              : 2017:06:16 15:11:06
Create Date                     : 2017:06:16 15:11:06
Components Configuration        : -, Cr, Cb, Y
Compressed Bits Per Pixel       : 3.275035748
Shutter Speed Value             : 1/250
Aperture Value                  : 5.0
Exposure Compensation           : 0
Max Aperture Value              : 2.8
Subject Distance                : 0 m
Metering Mode                   : Average
Light Source                    : Daylight
Flash                           : No flash function
Focal Length                    : 8.8 mm
Warning                         : [minor] Possibly incorrect maker notes offsets (fix by 1783?)
Make                            : DJI
Speed X                         : +2.30
Speed Y                         : +0.10
Speed Z                         : +0.00
Pitch                           : -3.00
Yaw                             : -0.10
Roll                            : +4.30
Camera Pitch                    : -89.90
Camera Yaw                      : -1.70
Camera Roll                     : +0.00
Flashpix Version                : 0010
Color Space                     : sRGB
Exif Image Width                : 5472
Exif Image Height               : 3648
Interoperability Index          : R98 - DCF basic file (sRGB)
Interoperability Version        : 0100
Exposure Index                  : undef
File Source                     : Digital Camera
Scene Type                      : Directly photographed
Custom Rendered                 : Normal
Exposure Mode                   : Auto
White Balance                   : Manual
[... 44 lines cut ...]
Image UID List                  : (Binary data 66 bytes, use -b option to extract)
Total Frames                    : 1
Image Width                     : 5472
Image Height                    : 3648
Encoding Process                : Baseline DCT, Huffman coding
Bits Per Sample                 : 8
Color Components                : 3
Y Cb Cr Sub Sampling            : YCbCr4:2:2 (2 1)
Aperture                        : 5.0
GPS Altitude                    : 806.4 m Above Sea Level
GPS Latitude                    : 16 deg 36' 20.88" S
GPS Longitude                   : 49 deg 15' 27.72" W
GPS Position                    : 16 deg 36' 20.88" S, 49 deg 15' 27.72" W
Image Size                      : 5472x3648
Preview Image                   : (Binary data 253812 bytes, use -b option to extract)
Megapixels                      : 20.0
Scale Factor To 35 mm Equivalent: 2.7
Shutter Speed                   : 1/240
Thumbnail Image                 : (Binary data 10177 bytes, use -b option to extract)
Circle Of Confusion             : 0.011 mm
Field Of View                   : 73.7 deg
Focal Length                    : 8.8 mm (35 mm equivalent: 24.0 mm)
Hyperfocal Distance             : 1.41 m
Light Value                     : 12.6
```

yjmenezes, 2017-06-19:
@kguidonimartins 

I have a small dataset, you could try it to check your ODM. ( git odm_br )
Do you mind to share your dataset ?
I saw you have   git repo too.

btw, do you know your camera sensor width in mm ?

kguidonimartins, 2017-06-19:
@yjmenezes not for now

yjmenezes, 2017-06-19:
it is ok, my thought was to make help easier and to supply the ODM team with P4 dataset. DJI camera has some issues related with rolling shutter effect.

flyandi, 2017-06-23:
I have the exact same issue. Happy to share my data set:

https://www.dropbox.com/s/d84i2fowy1avpt0/images.zip?dl=0

I am running this of the latest Docker Hub Image with command:

```
docker run -it --user root     -v $(pwd)/images:/code/images     -v $(pwd)/odm_orthophoto:/code/odm_orthophoto     -v $(pwd)/odm_texturing:/code/odm_texturing     --rm opendronemap/opendronemap:latest
```

Result:

```
2017-06-23 06:30:44,807 Cleaning depthmap for image DJI_0039.JPG
2017-06-23 06:30:45,810 Cleaning depthmap for image DJI_0014.JPG
2017-06-23 06:30:46,795 Cleaning depthmap for image DJI_0049.JPG
2017-06-23 06:30:47,821 Cleaning depthmap for image DJI_0051.JPG
2017-06-23 06:30:48,849 Merging depthmaps
Killed
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 128, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```

Any ideas?

dakotabenjamin, 2017-06-23:
@flyandi that looks like you ran out of memory. Use this tag: `--opensfm-processes n` where n is less than the total number of CPU cores you have.

yjmenezes, 2017-06-23:
@flyandi 

thank you for sharing your data.

I run under a QEMU vm with 23GB memory 4 cores and watch TOP and saw it takes 21GB.

0- do you know DJI FC220 sensor width in mm ?
1- how much memory (  real and swap)  do your have ?
2- do you have GCP ?

[image]

my command line:

./run.sh --project-path /mnt/  flyandl --force-focal 4.7 --force-ccd 6.20 --use-exif 

DJI FC200 is not defined inside   sensor_data.json, i use 6.20 mm

    "DJI FC300S": 6.16,
    "DJI FC300X": 6.2,
    "DJI FC350": 6.17,

My guest VM:
                                  system     Standard PC (i440FX + PIIX, 1996)
/0/400                            processor  AMD FX(tm)-4100 Quad-Core Processor
/0/401                            processor  AMD FX(tm)-4100 Quad-Core Processor
/0/402                            processor  AMD FX(tm)-4100 Quad-Core Processor
/0/403                            processor  AMD FX(tm)-4100 Quad-Core Processor
/0/1000                           memory     23GiB Memória do sistema

yjmenezes, 2017-06-23:
@flyandi 

I run PMVS

 ./run.sh --project-path /mnt/  flyandl --force-focal 4.7 --force-ccd 6.20 --use-pmvs --pmvs-min-images 2 --use-exif 
[image]

pmvs vs open
[image]
sfm

dakotabenjamin, 2017-12-11:
This should be fixed in #724

closed 2018-03-28 by dakotabenjamin