OpenDroneMap/ODM #609: Computing neighbors errors

winwinnh, 2017-06-22:
For some reason, I ran into this problem while running this specific flight. This flight contains 176 images, all geotagged. 

python /home/winwinnh/Desktop/OpenDroneMap/run.py  --use-exif --time --dem --build-overviews --project-path pathtoproject

2017-06-22 00:01:41,792 Computing neighbors
/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py:238: RuntimeWarning: invalid value encountered in arccos
  return np.arccos(a.dot(b) / (np.linalg.norm(a) * np.linalg.norm(b)))
2017-06-22 00:01:51,553 Computing depthmap for image 2017_0524_164936_206_geotag.JPG
2017-06-22 00:02:30,196 Computing depthmap for image 2017_0524_164724_132_geotag.JPG
2017-06-22 00:03:07,329 Computing depthmap for image 2017_0524_164603_086_geotag.JPG
2017-06-22 00:03:45,525 Computing depthmap for image 2017_0524_164735_138_geotag.JPG
2017-06-22 00:04:23,963 Computing depthmap for image 2017_0524_165145_278_geotag.JPG
2017-06-22 00:05:04,032 Computing depthmap for image 2017_0524_164542_074_geotag.JPG
2017-06-22 00:05:45,810 Computing depthmap for image 2017_0524_165055_250_geotag.JPG
2017-06-22 00:06:26,299 Computing depthmap for image 2017_0524_164553_080_geotag.JPG
^CTraceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
    dense.compute_depthmaps(data, graph, reconstructions[0])
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 31, in compute_depthmaps
    parallel_run(compute_depthmap, arguments, processes)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 47, in parallel_run
    return [function(arg) for arg in arguments]
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 68, in compute_depthmap
    depth, plane, score = de.compute_patch_match()
KeyboardInterrupt
Traceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/winwinnh/Desktop/OpenDroneMap/scripts/opensfm.py", line 128, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/winwinnh/Desktop/OpenDroneMap/opendm/system.py", line 26, in run
    raise Exception("Child was terminated by signal {}".format(-retcode))
Exception: Child was terminated by signal 2

dakotabenjamin, 2017-06-22:
It looks like it was just a warning. Can you try to run the thing through and see if you get a dense point cloud despite the warning?

winwinnh, 2017-06-22:
The process stopped everytime.

[INFO]    Resized 175 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[WARNING] Found a feature matching done progress file in: /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm/matching_done.txt
[WARNING] Found a valid OpenSfM tracks file in: /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm/tracks.csv
[WARNING] Found a valid OpenSfM reconstruction file in: /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.json
[WARNING] Found a valid OpenSfM meshed reconstruction file in: /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.meshed.json
[WARNING] Found a valid OpenSfM NVM reconstruction file in: /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.nvm
[DEBUG]   running PYTHONPATH=/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm undistort /media/winwinnh/Work_Storage/LIFT_10/Job_1/RED/FLIGHT_5_RED/opensfm
Traceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/undistort.py", line 23, in run
    self.undistort_images(reconstructions[0], data)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/undistort.py", line 28, in undistort_images
    image = data.image_as_array(shot.id)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 65, in image_as_array
    return cv2.imread(self.__image_file(image), IMREAD_COLOR)[:,:,::-1]  # Turn BGR to RGB
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 57, in __image_file
    return self.image_files[image]
KeyError: u'2017_0524_164553_080_geotag.JPG'
Traceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/winwinnh/Desktop/OpenDroneMap/scripts/opensfm.py", line 126, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/winwinnh/Desktop/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
winwinnh@winwinnhUbuntu:~/Desktop/OpenDroneMap$

winwinnh, 2017-06-22:
The process just stops every times.

[INFO]    Resized 175 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[WARNING] Found a feature matching done progress file in: /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm/matching_done.txt
[WARNING] Found a valid OpenSfM tracks file in: /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm/tracks.csv
[WARNING] Found a valid OpenSfM reconstruction file in: /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.json
[WARNING] Found a valid OpenSfM meshed reconstruction file in: /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.meshed.json
[WARNING] Found a valid OpenSfM NVM reconstruction file in: /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.nvm
[DEBUG]   running PYTHONPATH=/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm undistort /media/winwinnh/Work_Storage/Job/Job_1/RED/FLIGHT_5_RED/opensfm
Traceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/undistort.py", line 23, in run
    self.undistort_images(reconstructions[0], data)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/undistort.py", line 28, in undistort_images
    image = data.image_as_array(shot.id)
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 65, in image_as_array
    return cv2.imread(self.__image_file(image), IMREAD_COLOR)[:,:,::-1]  # Turn BGR to RGB
  File "/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 57, in __image_file
    return self.image_files[image]
KeyError: u'2017_0524_164553_080_geotag.JPG'
Traceback (most recent call last):
  File "/home/winwinnh/Desktop/OpenDroneMap/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/winwinnh/Desktop/OpenDroneMap/scripts/opensfm.py", line 126, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/winwinnh/Desktop/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
winwinnh@winwinnhUbuntu:~/Desktop/OpenDroneMap$

dakotabenjamin, 2017-06-22:
Delete the `opensfm` folder and try again.

winwinnh, 2017-06-22:
It didn't give that error after deleting opensfm then rerunning it. Also deleted this image cause I thought it was also the problem. 

2017-06-22 00:06:26,299 Computing depthmap for image 2017_0524_164553_080_geotag.JPG

The program completed normally after that, except it did not create the orthophoto.tif, only png. The png however is pretty messed up.

[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_texturing/odm_textured_model.obj
[DEBUG]   running /home/winwinnh/Desktop/OpenDroneMap/SuperBuild/install/bin/texrecon /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/opensfm/reconstruction.nvm /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_meshing/odm_mesh.ply /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_texturing/odm_textured_model -d gmi -o gauss_clamping -t none     
/home/winwinnh/Desktop/OpenDroneMap/SuperBuild/install/bin/texrecon (built on Jun 19 2017, 01:29:30)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 99987 verts... 199908 faces... done.
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 155
NVM: Number of features: 0
        Loading 100%... done. (Took 12.092s)                                    
Building adjacency graph: 
        Adding edges 100%... done. (Took 0.563s)                                
	299751 total edges.
View selection:
	Building BVH from 199908 faces... done. (Took: 373 ms)
        Calculating face qualities 100%... done. (Took 8.584s)                  
        Postprocessing face infos 100%... done. (Took 0.057s)                   
	Maximum quality of a face within an image: 21107.9
	Clamping qualities to 54.8859 within normalization.
	Writing data cost file... done.
	93591 faces have not been seen by a view.
	Comp	Iter	Energy		Runtime
	0000	0	232052	0.005
	0000	1	190052	0.14
	0000	2	172870	0.269
	0000	3	168730	0.398
	0000	4	163491	0.526
	0000	5	159771	0.655
	0000	6	155958	0.784
	0000	7	152373	0.913
	0000	8	148780	1.042
	0000	9	145036	1.171
	0000	10	141882	1.3
	0000	11	139464	1.433
	0000	12	137220	1.568
	0000	13	135216	1.705
	0000	14	133659	1.839
	0000	15	131885	1.972
	0000	16	130568	2.105
	0000	17	129711	2.238
	0000	18	128429	2.371
	0000	19	127737	2.504
	0000	20	126805	2.634
	0000	21	126111	2.764
	0000	22	125629	2.893
	0000	23	125085	3.023
	0000	24	124517	3.152
	0000	25	124194	3.281
	0000	26	123902	3.412
	0000	27	123275	3.541
	0000	28	122861	3.67
	0000	29	122712	3.798
	0000	30	122332	3.928
	0000	31	122112	4.056
	0000	32	121927	4.185
	0000	33	121846	4.313
	0000	34	121701	4.442
	0000	35	121683	4.57
	0000	36	121488	4.699
	0000	37	121319	4.829
	0000	38	121123	4.969
	0000	39	121073	5.109
	0000	40	120946	5.241
	0000	41	120973
	0000	Increase of energy - stopping optimization
	Took: 18.735s
Generating texture patches:
	Running... done. (Took 2.564s)
	3900 texture patches.
Running global seam leveling:
	Create matrices for optimization...  done.
	Lhs dimensionality: 74741 x 74741
	Calculating adjustments:
		Color channel 0: CG took 100 iterations. Residual is 9.50446e-05
		Color channel 1: CG took 100 iterations. Residual is 9.62947e-05
		Color channel 2: CG took 102 iterations. Residual is 9.82383e-05
		Took 0.221 seconds
        Adjusting texture patches 100%... done. (Took 1.16s)                    
Running local seam leveling:
        Blending texture patches 100%... done. (Took 13.6s)                     
Generating texture atlases:
	Sorting texture patches... done.
	Working on atlas 21 100%... done.
	Finalizing texture atlases... done. (Took: 5.727s)
Building objmodel:
	Saving model... done.
Whole texturing procedure took: 92.558s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_georeferencing/coords.txt
[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
Image is missing GPS Latitude data
For more detailed information, see log file.

[WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file. 
[WARNING] Found a valid georeferenced model in: /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_georeferencing/odm_georeferenced_model.ply
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[DEBUG]   running /home/winwinnh/Desktop/OpenDroneMap/build/bin/odm_orthophoto -inputFile /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_texturing/odm_textured_model.obj -logFile /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_orthophoto/odm_orthophoto_log.txt -outputFile /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCornerFile /media/winwinnh/Work_Storage/JOB/Job_1/RED/FLIGHT_5_RED/odm_orthophoto/odm_orthophoto_corners.txt
[WARNING] No coordinates file. A georeferenced raster will not be created
[INFO]    Running ODM OrthoPhoto Cell - Finished
[INFO]    OpenDroneMap app finished - Thu Jun 22 17:25:17  2017
winwinnh@winwinnhUbuntu:~/Desktop/OpenDroneMap$ 

[image]

dakotabenjamin, 2017-06-23:
Do your images ahve GPS in their EXIF? Otherwise you will need to either geotag them if you have the drone GPS track (most mission planning software has this functionality) or [generate a GCP file](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Running-OpenDroneMap#running-odm-with-ground-control).

winwinnh, 2017-06-25:
They do.
We had 6 flights total, 4 of them worked perfectly and the other 2 does the above. For the one that worked, the images came out with randomly distributed small/missing dots/areas and distorted around the edges.
Had a buddy of mine try the free version of Pix4D, and all the individual flights worked perfectly.

dakotabenjamin, 2017-06-28:
Some possible things that I can suggest, but I don't know without seeing the data.

1. Something is wrong with the image GPS (malformed, missing, or otherwise)
2. The reconstruction failed to create a cohesive reconstruction. This can happen if the overlap was low or there are a lot of difficult-to-reconstruct features like dense vegetation. 

I think ODM is a little more finicky about data quality than pix4d. Something for us to work on and improve for sure. 

I can try to offer some ways to hopefully fix the problems. 

1. check the image exif with a tool like exiftool to see if there's anything wrong or missing
2. Try these parameters: 
```
--resize-to 2000
if you make the images smaller, it can improve the reconstruction, at the expense of some quality in the mosaic
--min-num-features 8000
increasing the number of features could potentially enable a complete reconstruction
```

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/computing-neighbors-errors/390

closed 2018-03-28 by dakotabenjamin