OpenDroneMap/ODM #610: Install errors: not found GDAL

bryanibit, 2017-06-23:
FindGDAL.cmake:
 gdal-config not found. Please set it manually: GDAL_CONFIG=GDAL_CONFIG-NOTFOUND
-- Could NOT find GDAL (missing:  GDAL_LIBRARY GDAL_INCLUDE_DIR) (Required is at least version "1.9.0")
CMake Error at cmake/gdal.cmake:12 (message):
  GDAL support is required
Call Stack (most recent call first):
  CMakeLists.txt:108 (include)

-- Configuring incomplete, errors occurred!

The above is the error info. Do i download and install GDAL manually? Are there anyone to help me? Thank you!

dakotabenjamin, 2017-06-23:
gdal should install in the configure script. If running `./configure.sh reinstall` doesn't work, you can manually install with `sudo apt-get install libgdal-dev`

bryanibit, 2017-06-23:
Ok, I will try it manually.

closed 2017-06-24 by bryanibit