OpenDroneMap/ODM #611: images_resize written coordinates are wrong.

dakotabenjamin, 2017-06-23:
Original image exiftool out: 
```
GPS Version ID                  : 3.2.0.0
GPS Latitude Ref                : North
GPS Latitude                    : 41 deg 32' 56.76"
GPS Longitude Ref               : West
GPS Longitude                   : 83 deg 41' 56.31"
GPS Altitude Ref                : Above Sea Level
GPS Altitude                    : 122.3 m

```

images_resize of the same images:
```
GPS Version ID                  : 3.2.0.0
GPS Latitude Ref                : North
GPS Latitude                    : 4603233 deg 4' 18.19"
GPS Longitude Ref               : East
GPS Longitude                   : 274896 deg 11' 35.11"
GPS Altitude Ref                : Above Sea Level
GPS Altitude                    : 12230 m
```

yjmenezes, 2017-06-23:
@dakotabenjamin 

it is writing UTM.

cat << EOF  | tr -d [:blank:] | sed s/deg/d/g | tr \, \\t | awk -F \\t '{ print $2,$1 }' | \
    proj  -f "%.3f" +proj=utm +zone=17 +north +ellps=WGS84 
41 deg 32' 56.76" N, 83 deg 41' 56.31" W
EOF

output:
274896.144	4603233.203

dakotabenjamin, 2017-06-23:
They should be in lat-long, but that seems like a trivial fix.

dakotabenjamin, 2017-09-01:
The resize step has been removed in #644, rendering this more or less fixed

closed 2017-09-01 by dakotabenjamin