OpenDroneMap/ODM #613: Struck in the process of [ 85%]Built target pdal

bryanibit, 2017-06-24:
[ 84%] Performing install step for 'pdal'
[  2%] Built target pdal_jsoncpp
[  4%] Built target pdal_arbiter
[  6%] Built target dimbuilder
[  7%] Built target generate_dimension_hpp
[ 13%] Built target pdal_boost
[ 18%] Built target pdal_util
[ 97%] Built target pdal_base
[100%] Built target pdal_plugin_reader_pgpointcloud
[100%] Built target pdal
[100%] Built target pdal_plugin_writer_pgpointcloud
Install the project...
......

-- Installing: /home/inin/OpenDroneMap/SuperBuild/install/bin/pdal
-- Set runtime path of "/home/inin/OpenDroneMap/SuperBuild/install/bin/pdal" to "/home/inin/OpenDroneMap/SuperBuild/install/lib"
-- Installing: /home/inin/OpenDroneMap/SuperBuild/install/lib/pkgconfig/pdal.pc
-- Installing: /home/inin/OpenDroneMap/SuperBuild/install/bin/pdal-config
[ 85%] Completed 'pdal'
[ 85%] Built target pdal

I was struck in the above for more than one hour. I used 'install' and 'reinstall', but the result is the same like above. I do not know how to do. The computer has 8GB memory and Intel® Core™ i7-6700 CPU @ 3.40GHz × 8 CPU with Ubuntu 14.04. I don't think it should take so long. Is there somebody who can tell me the reason? Thank you!

bryanibit, 2017-06-24:
`list(APPEND CMAKE_MODULE_PATH "${CMAKE_CURRENT_LIST_DIR}/cmake/Modules/")`
I add the above to odm_25dmeshing/CMakeLists.txt and copy FindEigen3.cmake to the new folder "${CMAKE_CURRENT_LIST_DIR}/cmake/Modules/". Then anything seems like fine. Who can explain this?

grizzlyau, 2017-06-25:
What operating system are you using?

bryanibit, 2017-06-25:
Ubuntu 14.04

JakeC2i, 2017-06-27:
I'm getting the same error.

Can you tell me where exactly in odm_25dmeshing/CMakeLists.txt did you add that line, and where to copy the FindEigen3.cmake (I don't know how to understand the "${CMAKE_CURRENT_LIST_DIR}/cmake/Modules/", I guess CMAKE_CURRENT_LIST_DIR is a variable...).

I'm new to Ubuntu in general so I'm sorry if this is a noob question. I'm trying to setup odm_vagrant on virtual box  in Windows for almost 3 days now with no luck, and this is the furthest I've come.

yjmenezes, 2017-06-27:
@bryanibit 

I compile and run ODM v 0.3.1, today's commit and configure.sh install  works fine for me.
QEMU vm under Debian GNU/Linux 8.8x64 host and Ubuntu 16.04.2 guest.
Maybe you should try Ubunu 16.04.2 LTS.

bryanibit, 2017-06-28:
@JakeC2i  I add 'list(APPEND CMAKE_MODULE_PATH "${CMAKE_CURRENT_LIST_DIR}/cmake/Modules/")' under 'if (eigen_found)'. At the same time, I put FindEigen.cmake file to '${CMAKE_CURRENT_LIST_DIR}/cmake/Modules/' folder and ${CMAKE_CURRENT_LIST_DIR} variable means the current address of CMakeLists.txt file. So renew /cmake/Modules folder including FindEigen.cmake.
If not working, I think you could read [Find Libraries CMakeLists.txt](https://cmake.org/Wiki/CMake:How_To_Find_Libraries#Using_external_libraries_that_CMake_doesn.27t_yet_have_modules_for
). In effect, the Eigen need not to be compiled but pointing their header, namely, INCLUDE_DIRS variable.

bryanibit, 2017-06-28:
@yjmenezes Thank you for your advices. I would try 16.04 LTS one day.

JakeC2i, 2017-06-28:
@bryanibit Thank you very much, I'll try it later today.

NikhilMadaan, 2017-07-05:
@bryanibit sir, 
why are we adding FIndEigen.cmake rather than FIndEigen3.cmake beacause that is the file which is creating this error?
Please help i'm confused where to put which file.

Thanks in advance

bryanibit, 2017-07-05:
@NikhilMadaan  I think you should make sure the version of ODM you used. Is it the lastet one? Eigen 3 is used now. The file put the place CMakeLists.txt points. The original file (FindEigen3.cmake) is in OpenDroneMap/modules/odm_25dmeshing. I just copy the file to OpenDroneMap/modules/odm_25dmeshing/cmake/Modules and add the path to CMakeLists.txt.

NikhilMadaan, 2017-07-06:
yes i'm on the latest release and the problem got solved by setting the path by using 
`set` Eigen_DIR=` followed by the path

naivefun, 2017-07-06:
@bryanibit  I'm using the vagrant version and see the same error. Shall we update the vagrant box?

facundocastello, 2017-07-06:
I've the same error and i solve it adding a path that have findEigen3.cmake to the CMAKE_MODULE_PATH of the CMakeLists.txt
i did the next:
Old line:
`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_SOURCE_DIR} )`
New line:
`set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_SOURCE_DIR} /your/path/OpenDroneMap/modules/)`

**this path must be any folder that contains "FindEigen3.cmake"**

bryanibit, 2017-07-07:
@naivefun I did not use ODM on vagrant. I used it on VMware and real PC. I think you can try 'configure.sh reinstall'.

bryanibit, 2017-07-07:
@NikhilMadaan  @facundocastello  In effect, 'find_package' works like the two mode:

1. CMAKE checks all directories in ${CMAKE_MODULE_PATH}
   then it looks its own directory <CMAKE_Root>/share/cmake-x.y/Modules
   <CMAKE_Root> is usually /usr/

2. If no such file is found, it looks for <NAME>Config.cmake e.g. OpenCVConfig.cmake
   or <low-case-name>-config.cmake 

   no matter what mode is used, the following will be defined:
```
   <NAME>_FOUND
   <NAME>_INCLUDE_DIRS or <NAME>_INCLUDES
   <NAME>_LIBRARIES or <NAME>_LIBS
   <NAME>_DEFINITIONS
```
_All this_ takes places in the Find<NAME>.cmake file

Hense, for the second mode, you can set (Eigen_DIR 'variable') or set ${CMAKE_MODULE_PATH} including .cmake

dakotabenjamin, 2017-07-17:
@pierotofy can you make sense of this? Is there something we should be fixing in 25dmeshing?

pierotofy, 2017-07-17:
It's odd that this problem only happens on certain versions of Ubuntu (16.04 LTS works fine, as the docker images and other users have confirmed).

Maybe explicitly setting the path to FindEigen3.cmake here https://github.com/OpenDroneMap/OpenDroneMap/blob/master/modules/odm_25dmeshing/CMakeLists.txt#L4 would correct the compilation on other platforms, perhaps somebody can try and confirm.

There should be no need to move files around.

dakotabenjamin, 2017-07-18:
I can confirm the OP issue

Ekci, 2017-08-10:
Any update? I am trying to build ODM on 14.04 and can't find a way to fix Eiden3 problem.

bryanibit, 2017-08-11:
@Ekci I think facundocastello's method will work. If not, you can try what I did.

pierotofy, 2020-09-11:
We've made some major changes to the build process, which should fix these errors.

If not, please re-open? :pray:

closed 2020-09-11 by pierotofy