OpenDroneMap/ODM #614: The repository 'http://ppa.launchpad.net/george-edison55/cmake-3.x/ubuntu zesty Release' does not have a Release file

grizzlyau, 2017-06-25:
OS:  Ubuntu 17.04.  Brand new install running as a VirtualBox VM.  

There appears to be a problem with the cmake-3.x repo, it does not download properly.  As a result, ODM does not compile properly.  

After running 'configure.sh install' from the ODM directory, there were several errors related to make. 

I tried to run an update 'apt-get update' then got the errors below from cmake-3.x

*W: The repository 'http://ppa.launchpad.net/george-edison55/cmake-3.x/ubuntu zesty Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: The repository 'http://ppa.launchpad.net/ubuntugis/ppa/ubuntu zesty Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://ppa.launchpad.net/george-edison55/cmake-3.x/ubuntu/dists/zesty/main/binary-i386/Packages  404  Not Found
E: Failed to fetch http://ppa.launchpad.net/ubuntugis/ppa/ubuntu/dists/zesty/main/binary-i386/Packages  404  Not Found
E: Some index files failed to download. They have been ignored, or old ones used instead.*

I am looking for the error logs from the install, I might have to run the install again to take note of where they are as I forget.

grizzlyau, 2017-06-25:
Attaching CMakeOutput.log as a txt file as github does not support .log file format upload.  

[CMakeOutput.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1100094/CMakeOutput.txt)

grizzlyau, 2017-06-25:
When I look at the CMake 3.x project page, they list support for Ubuntu 14.04, 15.10 and 16.04 but NOT 17.04.  I wonder if this means ODM does not support 17.04?? 

From https://launchpad.net/~george-edison55/+archive/ubuntu/cmake-3.x

*CMake 3.x
PPA description

Provides up to date builds of CMake 3.x (and related packages) for recent releases of Ubuntu. This currently includes:

* 14.04 - Trusty Tahr
* 15.10 - Wily Werewolf
* 16.04 - Xenial Xerus

All packages are built for i386, amd64, arm64, and armhf.

These builds were created by taking the packages for CMake 3.x from Debian releases and backporting them and are considered stable.*

grizzlyau, 2017-06-25:
Found out that Ubuntu 17.04 is not supported.  Please please update Installation Wiki to let people know.  Would have saved me many hours.

grizzlyau, 2017-06-25:
Got an .iso for Ubuntu 16.04.2 and it installed without issues in a Virtual Box VM.  It took a very long time, many hours, to download, install, compile etc.  

As an experiment I fired up a Google Cloud VM of Ubuntu 16.04.2 and am currently running the install on that.  The difference in speed is incredible, it is flying through the install and compile process at the moment.

dakotabenjamin, 2017-06-29:
Sorry you ran into trouble! I'm working on getting the changes necessary to support current versions of Ubuntu but there are some big breaking changes. I updated the wiki. BTW, anyone can update the wiki, in the future if you wish to contribute as well.

closed 2017-06-29 by dakotabenjamin