OpenDroneMap/ODM #616: Importing OpenSfM dense reconstruction using Docker

dgriffiths3, 2017-06-26:
Hi there,

Is is possible to process the OpenSfM point cloud independently, then import the results into ODM to geo-reference and create orthophoto? I am on MacOS and running ODM using docker.

Any help would be very appreciated,
Dave

dakotabenjamin, 2017-06-29:
I'm hoping we can get the opensfm updates synched up with ODM in the next few weeks. but I think you could try to run odm up to opensfm (try the `--end-with resize` param), run opensfm, copy the results into an opensfm dir in your odm project, then run odm again (it will not overwrite if it is able to find the proper outputs). Theoretically it would work. 

This brings up an interesting point regarding importing/exporting. Something to ponder.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/importing-opensfm-dense-reconstruction-using-docker/373

closed 2018-03-28 by dakotabenjamin