OpenDroneMap/ODM #618: failure to start up container with custom instructions

NyakudyaA, 2017-06-30:
# Problem

When running the following command

`docker run -it --rm -v $(pwd)/images:/code/images -v $(pwd)/odm_orthophoto:/code/odm_orthophoto -v $(pwd)/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap `

With data coming from [data](https://github.com/OpenDroneMap/odm_data_copr)

```
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/code/opensfm/reconstruction.json'
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 112, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

dakotabenjamin, 2017-07-05:
I ran that dataset with that run command with no errors during the opensfm step. cna you try a different dataset like [toledo](https://github.com/OpenDroneMap/odm_data_toledo) to see if it's the dataset or something else?

dakotabenjamin, 2017-07-17:
@NyakudyaA any luck?

NyakudyaA, 2017-07-20:
I haven't got to test it. Will comment later during the day

NyakudyaA, 2017-07-20:
@dakotabenjamin I have changed the dataset and it's working fine

closed 2017-07-20 by NyakudyaA