OpenDroneMap/ODM #619: Fresh Ubuntu 16.04 build: What to do next?

Harsha2398, 2017-07-01:
I'm a newbie to Ubuntu. I've been familiarising myself with the terminal and I followed the instructions for installing ODM 0.3.1 beta, but the install did not finish completely. Some packages were not found apparently, and this was on 17.04.
I'm planning on going back to 16.04 and I wish to start from the clean install stage itself: Will opendronemap take care of all it's dependencies or would I have to do setup somethings on my own? (after installing git of course)

sukoshi1507, 2017-07-01:
Hello,

Read the instructions,  i managed to install ODM using all instructions / scripts provided.

Harsha2398, 2017-07-01:
@sukoshi1507 @dakotabenjamin 

I’ve recently installed Ubuntu 16.04.02 LTS and right from the clean install stage, after installing CMake, I cloned your OpenDroneMap Beta 0.3.1 project and ran the install. Although it did say configuration complete (there were a few errors that were noticed when dependencies were being installed, it’s too late for me to provide you a log now), but when I ran the file I got this error:  ImportError: libecto.so.0.6: Directory not found.

This are the environment variables I’ve put in the .bashrc file 
(export PYTHONPATH=$PYTHONPATH:/home/sriharsha/downloads/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export PYTHONPATH=$PYTHONPATH:/home/sriharsha/downloads/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/sriharsha/downloads/OpenDroneMap/SuperBuild/install/lib).

 Have I made a mistake somewhere? **What do I need to change in my existing setup so that I can make it work?**

What if I was to start from scratch, again (from a previous snapshot), again from a clean install, what all would I have to install before running your configure.sh file, so that everything will be smoothly installed?

Harsha2398, 2017-07-01:
@sukoshi1507 Hey, I was able to install it too, while running the file, I got the error I mentioned above

yjmenezes, 2017-07-01:
@Harsha2398 

i have compiled and run ODM v 0.3.1 under Ubuntu 16.04.2 LTS and all is fine for me.
after install git-core i clone  ODM and compile and run from inside OpenDroneMap as $ user.
1- try log your compilation messages:
./configure.sh install   2> /tmp/error.txt

2- i use run.sh. it set Python path. add some extra parameter if necessary, like this:
./run.sh --project-path  /mnt/   project_name_directory   --force-focal 4.7 --force-ccd 6.20 --use-exif 

please take a look  issue # 601

Harsha2398, 2017-07-03:
[Configuration done, logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119109/Configuration.done.logs.txt)
[ecto logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119110/ecto.logs.txt)
[errorfile generated .txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119108/errorfile.generated.txt)
[mvstexturing to pcl logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119111/mvstexturing.to.pcl.logs.txt)
[OpenCV and onwards_ Logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119112/OpenCV.and.onwards_.Logs.txt)
[PCL logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119113/PCL.logs.txt)
[PDAL logs.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1119114/PDAL.logs.txt)

Here are the logs I managed to collect from my clean Ubuntu 16.04.2 LTS VM.

@sukoshi1507 @yjmenezes @dakotabenjamin  Please take a look and tell me what I should do to avoid them or if they will affect your program from running

dakotabenjamin, 2017-07-05:
There are no errors in the logs. Did you try @yjmenezes suggestions to use run.sh? That should solve your issue.

Harsha2398, 2017-07-05:
I will soon and update you on it. But i noticed some features were disabled, what is all that about?

The following subsystems will be built:
--   common
--   octree
--   io
--   kdtree
--   search
--   surface
-- The following subsystems will not be built:
--   geometry: Disabled manually.
--   sample_consensus: Disabled manually.
--   filters: Disabled manually.
--   features: Disabled manually.
--   segmentation: Disabled manually.
--   visualization: Disabled manually.
--   registration: Disabled manually.
--   keypoints: Disabled manually.
--   tracking: Disabled manually.
--   recognition: Disabled manually.
--   apps: OpenNI was not found.
--   outofcore: Disabled manually.
--   examples: Code examples are disabled by default.
--   people: Disabled manually.
--   global_tests: No reason
--   tools: Disabled manually.

I mean this. It won't hamper anything while running? Right?

dakotabenjamin, 2017-07-05:
They are not necessary for running ODM, so we disable those to keep the install as small as we can.

Harsha2398, 2017-07-05:
I tried running this line and got too few arguments as an error
`./run.sh --project-path ~/Pictures/ODMProjects/odm_data_bellus/images
`
Is there a mistake in my syntax? I noticed my issue was similar to #365 at the ending side of his comments. So I followed what you told Bavar2142 there! Sorry if this is all to silly to you. 

P.S: Out of curiosity, how can we enable those files you mentioned above?

dakotabenjamin, 2017-07-05:
The command line has changed a little since then, that would lead to confusion. Try: `./run.sh --project-path ~/Pictures/ODMProjects odm_data_bellus`

You can see [here](https://github.com/OpenDroneMap/OpenDroneMap/wiki/Run-Time-Parameters) a more detailed syntax explanation.

Harsha2398, 2017-07-06:
Thanks @sukoshi1507, @yjmenezesm, @dakotabenjamin, for your patience, guys! It worked. Thank you very much again. Sorry for all the noob mistakes haha!

P.S: @dakotabenjamin I have a few more doubts regarding the installations, is it okay if I email you?

closed 2017-07-06 by Harsha2398