OpenDroneMap/ODM #621: fixing calibration and projection matrix

Gregoriomartjim, 2017-07-05:
Hello 

Is there a way to fix the calibration and projection matrix? i already have this parameters from my cam , so i would like to use this parameters in ODM, the real problem is that i would like to make many different maps but i want them all to begin in the same orientations (rotation and translation) so i can attach all point clouds and maps at the end.

Thanks and Regards

dakotabenjamin, 2017-07-05:
Hey, check out this and see if it works for you: https://github.com/dakotabenjamin/CameraCalibration

Gregoriomartjim, 2017-07-05:
Thanks for the quick answer, i already have a way to get the calibration matrix and the projection matrix, basically doing the same as your code, but i would like to fix this parameters in OpenDroneMap, so i can have all the same parameters for all my maps. 

Cheers

dakotabenjamin, 2017-07-05:
I'm not sure what you are asking, then. Can you clarify, or provide examples?

Gregoriomartjim, 2017-07-05:
Sure, when i run ODM for two different set of pictures i get pointclouds , but the orientation of these point clouds is different (they are rotated and with a different origin) even when the pictures where acquired using the same camera (i already have the parameters of this camera , projection and camera matrix). I think if i can ask ODM to use the same parameters always the generated point clouds are going to have the same orientation. Thas why i would like to use my camera parameters, so i will generate point clouds with the same orientation. The idea is to stitch this point clouds later, that is why is important to have the same orientations.

dakotabenjamin, 2017-07-05:
The point cloud orientation AFAIK would be unaffected. Are you using the georeferenced point cloud? That would be reoriented and able to be merged with other referenced clouds within some error margin. 

Currently we're working on some scaling improvements that might be of relevance: https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/

It's still a WIP but hope to have an update soon.

Gregoriomartjim, 2017-07-05:
The orientation changes always when i use two different set of images even coming from the same camera i have test this many times. No i am not using the georeference point cloud i could try with this, but however i wouldn't like to use this option. I check the sensor_data_detailed.json, but is there a way to know which one is using.?

KommandorKeen, 2017-07-26:
If you are not using the georeferenced cloud then the cloud will arbitrarily be aligned. Struggling to se the issue. I am finding good agreement between georefed clouds

Gregoriomartjim, 2017-07-26:
Hello @KommandorKeen  could you please send me your picture so i can make a try, i had use even the georeferenced cloud but i still dont have a good match after processing the images with ODM. check this link where we have been discussing the problem https://github.com/OpenDroneMap/OpenDroneMap/issues/623#issuecomment-317562095

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/fixing-calibration-and-projection-matrix/367

closed 2018-03-28 by dakotabenjamin