OpenDroneMap/ODM #623: aligning two point clouds using georeference

Gregoriomartjim, 2017-07-06:
Hello 

I have some photos with georeference. i used opendronemap to get the point cloud but i cant get the point clouds aligned, (i used the altitude, latitude and longitude) however i cant manage to align them. 

Is there a way to do this or am i doing something wrong?
Is there a way to set the origin in opendronemap?

this is a picture of the 2 point cluods
[image]

Regards

dakotabenjamin, 2017-07-11:
A naive alignment will not likely work in this case. ODM currently does not have this functionality. So are these two point clouds encompassing the same area? I'm not sure exactly what I'm looking at

Gregoriomartjim, 2017-07-11:
yes they are the same area, but different pictures. They look to have the same plane orientation, but one is above the other.

dakotabenjamin, 2017-07-11:
OK, I think this is outside the scope of ODM, so I will close this issue.

yjmenezes, 2017-07-11:
@dakotabenjamin , i apologise, let me reopen this issue.

@Gregoriomartjim 

it is interesting, to compare same area at different time. ( temporal analysis )
Could you, please, share your command line options and your  "logs" , and  gcp_list.txt, if available ?

Gregoriomartjim, 2017-07-11:
well i got the GPS coordinates from a different sensor, then using exiftool i add the coordinates to the pictures, finally i used this in ODM. how could i upload the pictures ?

yjmenezes, 2017-07-11:
@Gregoriomartjim 

you can create a new repo and upload your images.
Git has a 25MB limit for upload.
compress your images using several blocks if necessary.

Gregoriomartjim, 2017-07-11:
https://github.com/Gregoriomartjim/gps-pictures ok i put the images here, i use the first 5 for one map and the next 5 for the second map, thanks for the answers

yjmenezes, 2017-07-12:
@Gregoriomartjim 

It seems you do not have a good dataset.
please check:
1- exif coordinates, two images at same location
2- image size  856x480
3- camera focal mm ??? sensor width mm ???
[odm_gregorio.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1142326/odm_gregorio.zip)
4- few images,  two overlapping strips would be better.
5- GSD or flight height ?

what parameters are you using at command line ?

a ground gcp_list.txt points could  help

Gregoriomartjim, 2017-07-14:
Hello i download a set of pictures with all the required data   [https://github.com/Gregoriomartjim/gpsimages-full-/upload/master](url), however i still have the same problem, i used the first ten in one folder and the next ten in the other, then ODM and finally i open them with meshlab, they both are in the same plane but they are stacked up, i would like to find a way to stitch the point clouds in the corresponding part in the map.  

Here is how the point clouds look , the parking lot is stacking up in the house part
[image]

this is the map , check out the parking lot and house.
[image]

yjmenezes, 2017-07-17:
@Gregoriomartjim 

I have a good dataset with painted targets gcps.  Two sets, same location, same gcps.
i am asking data set owner for permission to share  with ODM list or thru my git repo.

Gregoriomartjim, 2017-07-17:
Thanks for your help

dakotabenjamin, 2017-07-17:
So I think I know what is going on here. Our version of OpenSfM doesn't use altitude tags by default (OpenSfM is typically used for ground level photos from mobile phone cameras which have bad altitude values) but this could be enabled when we update OpenSfM. I'm hoping to get that working this week. If you're willing to test let me know.

Gregoriomartjim, 2017-07-17:
yes i am totally willing to help you guys. i am working in a project where i will have to use many of the tools you have developed, so i am really interested in learn how to use OpenSfM and ODM

yjmenezes, 2017-07-19:
@GeospatialDaryl 

if you like, i have this small and good dataset. two sets, same location and gcps.
  | set02a.tar.bz2 | 2017-07-16 18:10 | 94M |  
  | set02b.tar.bz2 | 2017-07-16 18:10 | 89M |  
  | set02gcp.tar.bz2 | 2017-07-16 18:35 | 5.9M |  

i will let it available for you here:
http://cartognu.org/hakan/

please notify me if you download it.

Gregoriomartjim, 2017-07-19:
Thanks yjmenezes i will try to do it and post the results 

regards

Gregoriomartjim, 2017-07-19:
the maps are in the same plane however now we can notice a rotation , even with all the coordinates , i understand i could merge them in different manners but i would like to generate the point clouds already with the same orientations. 
[image]

yjmenezes, 2017-07-20:
@Gregoriomartjim 

please, could you describe your steps ?
georeferencing is the ODM last stage ( final pipiline stage ),  models are in local coordinate system.
take a look odm_georeferencing_log.txt

Gregoriomartjim, 2017-07-20:
I im just running ODM and then plotting the point clouds together using meshlab, where is the odm_georeferencing_log.txt?

smathermather, 2017-07-21:
@Gregoriomartjim -- we are working on a branch specifically to split and rejoin datasets while using matching to ensure alignment:
https://github.com/paulinus/OpenDroneMap/tree/feature-split-dataset/scripts/metadataset

There's not a lot of documentation for this yet, but if you want to give it a try, it may help with what you're trying to do.

Gregoriomartjim, 2017-07-21:
@smathermather  thanks for the info but i would like to have them already in position from the begging, so i wont need to use other scritp. However thanks i will check it 

Regards

yjmenezes, 2017-07-23:
@Gregoriomartjim 

Are my dataset set02 useful for you ?
I have the set01, same location, overlapping the set02 at east side.
I think the idea can be usefull to break very large dataset into overlapping blocks.

Gregoriomartjim, 2017-07-24:
i just got the set02a and b , if you post your set01 i will try to put them together , i also will try to split one of the sets and try to overlap both of them.

cheers

yjmenezes, 2017-07-24:
@Gregoriomartjim 

Please, take odm_set01*  images + gcps from my server.
http://cartognu.org/hakan/

cheers

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/aligning-two-point-clouds-using-georeference/370

closed 2018-03-28 by dakotabenjamin