OpenDroneMap/ODM #624: ODM ImportError: No module named 'yaml'

dl2ecc, 2017-07-08:
Hi there,

I'm tiying to get the new version of ODM running.
I'm using Ubuntu 14.04 LTS in it's updated version.
Python version is 2.7.
I downloaded OpenDroneMap-OpenDroneMap-v0.3.1-61-gc43366a.zip and unziped it
in a folder named ODM3. Then I started ~/ODM3$ ./configure.sh install.
The script finished correctly.
After running python run.py ~~project-path /tests/ I got an
ImportError: No module named 'yaml'.
I installed pip but nothing changed.

Can you help me ?

Thanks
Regards
Wolfgang

yjmenezes, 2017-07-08:
@dl2ecc 

you could try:
sudo apt-get update;   sudo apt-get install python-yaml
and run thru shell script from OpenDromeMap directory.
./run.sh   

For me,. ODM v0.3.1 is working well under Ubuntu 16.04.2 LTS

dl2ecc, 2017-07-09:
Hi,
thanks for the help.
I installed yaml and I could go a little further.
Now I get ImportError: No module named opensfm.exif
May be there is a little work to get ODM3 working.

yjmenezes, 2017-07-09:
@dl2ecc 

you are welcome.
please, consider upgrade to 16.04 ( xenial ), to get more fresh dependencies solved.

dl2ecc, 2017-07-09:
Hi Julio-Cesar

thanks for your hint.
I have to make a backup from my system first and then I'm going 
to update my system.

Best Regards
Wolfgang

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/odm-importerror-no-module-named-yaml/386

closed 2018-03-28 by dakotabenjamin