OpenDroneMap/ODM #626: Single core used in Docker

eyeNsky, 2017-07-14:
Haven't check the project out in a while, looking good!

I ran the Docker workflow and it is only using one core on the host.

Settings:
Linux Docker 17.05.0-ce
docker system info -> CPUs: 8
docker run --cpuset-cpus has no impact.
--opensfm-processes has no impact.
gdal_translate -co NUM_THREADS=ALL_CPUS step also shows a single thread.

pierotofy, 2017-07-14:
Hi, welcome back 😄 

Setting `--opensfm-processes` should use multiple cores.

How are you checking for the core usage?

eyeNsky, 2017-07-14:
System Monitor on the host.

pierotofy, 2017-07-14:
Mm, this is what I'm currently getting using opensfm-processes 2 for example:

[image]

```
Docker version 17.05.0-ce, build 89658bed64
```

```
Kernel Version: 4.11.5-1-ARCH
Operating System: Arch Linux
OSType: linux
Architecture: x86_64
CPUs: 4
Total Memory: 23.42GiB
```

How does the CPU usage look like when you run:

```
docker run --rm -it --cpuset-cpus="0-3" jess/stress --cpu 4
```

?

eyeNsky, 2017-07-14:
That lights up 4 cores.

pierotofy, 2017-07-14:
Could you share the exact command you are using to run opendronemap?

eyeNsky, 2017-07-14:
Ahh, I was using --cpuset-cpus incorrectly (and unnecessarily!).

I had  --cpuset-cpus="3", but the "0-3" syntax works. However, it is not needed.  Setting --opensfm-processes has the desired effect.

Thanks and mark it "Closed: User error" !!

pierotofy, 2017-07-14:
Glad you figured it out! 😄

closed 2017-07-17 by dakotabenjamin

mrgloom, 2017-08-24:
Also worked for me, here is full command:

`docker run --cpus 8 -it --rm -v  <images_path>:/code/images -v <odm_orthophoto_path>:/code/odm_orthophoto -v <odm_georeferencing_path>:/code/odm_georeferencing opendronemap/opendronemap --opensfm-processes 8`

But I'm not sure if `opensfm-processes` will parallelize all steps of pipeline, is it true?

dakotabenjamin, 2017-08-24:
the rest of the pipeline is parallelized automatically where parallel processes exist. IIRC that's only in the ortho generation step

pierotofy, 2017-08-24:
2.5D mesh generation also uses all available cores. Not sure about the rest.