OpenDroneMap/ODM #627: Unable to get ODM3 running. Error 2

dl2ecc, 2017-07-15:
Hi there,
I tried to install ODM3 now for several days.
First I used Ubuntu 14.04 LTS and got lots of errors.
Then I updated to 16.04 LTS and Python 3.5.2
I used git cloning and later I downloaded ODM3.zip. from Github.
I tried bash config.sh install and sudo bash config.sh install with the same result.

Here's a copy of the last parts of the config procedure.
-------------------------------------------------------------------------------------------------------------------
[ 75%] Linking CXX executable ../../bin/odm_texturing
/usr/bin/ld: warning: libboost_system.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_common.so, may conflict with libboost_system.so.1.58.0
/usr/bin/ld: warning: libboost_thread.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_common.so, may conflict with libboost_thread.so.1.58.0
/usr/bin/ld: warning: libboost_filesystem.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_filesystem.so.1.58.0
/usr/bin/ld: warning: libboost_iostreams.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_iostreams.so.1.58.0
/usr/bin/ld: warning: libboost_chrono.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_chrono.so.1.58.0
CMakeFiles/odm_texturing.dir/src/OdmTexturing.cpp.o: In Funktion `pcl::io::loadPLYFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PolygonMesh&)':
OdmTexturing.cpp:(.text._ZN3pcl2io11loadPLYFileERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_11PolygonMeshE[_ZN3pcl2io11loadPLYFileERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_11PolygonMeshE]+0x2f2): Nicht definierter Verweis auf `pcl::PLYReader::read(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PolygonMesh&, int)'
collect2: error: ld returned 1 exit status
modules/odm_texturing/CMakeFiles/odm_texturing.dir/build.make:310: die Regel für Ziel „bin/odm_texturing“ scheiterte
make[2]: *** [bin/odm_texturing] Fehler 1
CMakeFiles/Makefile2:338: die Regel für Ziel „modules/odm_texturing/CMakeFiles/odm_texturing.dir/all“ scheiterte
make[1]: *** [modules/odm_texturing/CMakeFiles/odm_texturing.dir/all] Fehler 2
[ 79%] Linking CXX executable ../../bin/odm_georef
/usr/bin/ld: warning: libboost_system.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_common.so, may conflict with libboost_system.so.1.58.0
/usr/bin/ld: warning: libboost_thread.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_common.so, may conflict with libboost_thread.so.1.58.0
/usr/bin/ld: warning: libboost_filesystem.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_filesystem.so.1.58.0
/usr/bin/ld: warning: libboost_iostreams.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_iostreams.so.1.58.0
/usr/bin/ld: warning: libboost_chrono.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/5/../../../../lib/libpcl_io.so, may conflict with libboost_chrono.so.1.58.0
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In Funktion `Georef::readHeader(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2&, Eigen::Matrix<float, 4, 1, 0, 4, 1>&, Eigen::Quaternion<float, 0>&, int&, int&, unsigned int&, int)':
Georef.cpp:(.text+0x7a67): Nicht definierter Verweis auf `pcl::MTLReader::read(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In Funktion `Georef::loadObjFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, pcl::TextureMesh&)':
Georef.cpp:(.text+0x9909): Nicht definierter Verweis auf `pcl::MTLReader::getMaterial(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In Funktion `pcl::PLYWriter::write(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&, bool)':
Georef.cpp:(.text._ZN3pcl9PLYWriter5writeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_14PCLPointCloud2ERKN5Eigen6MatrixIfLi4ELi1ELi0ELi4ELi1EEERKNSC_10QuaternionIfLi0EEEb[_ZN3pcl9PLYWriter5writeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_14PCLPointCloud2ERKN5Eigen6MatrixIfLi4ELi1ELi0ELi4ELi1EEERKNSC_10QuaternionIfLi0EEEb]+0x12): Nicht definierter Verweis auf `pcl::PLYWriter::writeASCII(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&, int, bool)'
Georef.cpp:(.text._ZN3pcl9PLYWriter5writeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_14PCLPointCloud2ERKN5Eigen6MatrixIfLi4ELi1ELi0ELi4ELi1EEERKNSC_10QuaternionIfLi0EEEb[_ZN3pcl9PLYWriter5writeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_14PCLPointCloud2ERKN5Eigen6MatrixIfLi4ELi1ELi0ELi4ELi1EEERKNSC_10QuaternionIfLi0EEEb]+0x27): Nicht definierter Verweis auf `pcl::PLYWriter::writeBinary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&, bool)'
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In Funktion `int pcl::PLYWriter::write<pcl::PointXYZRGBNormal>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PointCloud<pcl::PointXYZRGBNormal> const&, bool, bool)':
Georef.cpp:(.text._ZN3pcl9PLYWriter5writeINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_10PointCloudIT_EEbb[_ZN3pcl9PLYWriter5writeINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_10PointCloudIT_EEbb]+0x128): Nicht definierter Verweis auf `pcl::PLYWriter::writeASCII(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&, int, bool)'
Georef.cpp:(.text._ZN3pcl9PLYWriter5writeINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_10PointCloudIT_EEbb[_ZN3pcl9PLYWriter5writeINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_10PointCloudIT_EEbb]+0x1b0): Nicht definierter Verweis auf `pcl::PLYWriter::writeBinary(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2 const&, Eigen::Matrix<float, 4, 1, 0, 4, 1> const&, Eigen::Quaternion<float, 0> const&, bool)'
CMakeFiles/odm_georef.dir/src/Georef.cpp.o: In Funktion `int pcl::io::loadPLYFile<pcl::PointXYZRGBNormal>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PointCloud<pcl::PointXYZRGBNormal>&)':
Georef.cpp:(.text._ZN3pcl2io11loadPLYFileINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_10PointCloudIT_EE[_ZN3pcl2io11loadPLYFileINS_17PointXYZRGBNormalEEEiRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERNS_10PointCloudIT_EE]+0x36f): Nicht definierter Verweis auf `pcl::PLYReader::read(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, pcl::PCLPointCloud2&, Eigen::Matrix<float, 4, 1, 0, 4, 1>&, Eigen::Quaternion<float, 0>&, int&, int)'
collect2: error: ld returned 1 exit status
modules/odm_georef/CMakeFiles/odm_georef.dir/build.make:338: die Regel für Ziel „bin/odm_georef“ scheiterte
make[2]: *** [bin/odm_georef] Fehler 1
CMakeFiles/Makefile2:173: die Regel für Ziel „modules/odm_georef/CMakeFiles/odm_georef.dir/all“ scheiterte
make[1]: *** [modules/odm_georef/CMakeFiles/odm_georef.dir/all] Fehler 2
Makefile:83: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
Configuration Finished
-------------------------------------------------------------------------------------------------------------------
Can anyone help me ?
Thanks a lot.
Regards
Wolfgang

yjmenezes, 2017-07-15:
@dl2ecc 

try switch back to python2.7

# step-0 current version
python --version
# step-1 available
ls /usr/bin/python*
# step-2 change
alias python='/usr/bin/python2.7'
# step-3 relog
. ~/.bashrc
# step-4  check
python --version

yjmenezes, 2017-07-15:
@dl2ecc 
@dl2ecc 

I git ODM v0.3.1, today,   to my  home folder.
No problem to compile under Ubuntu 16.04.2  lsb-release.
I compile as unprivileged user.
cd  ~/OpenDroneMap
./configure.sh install 

all is fine. 

I am running Ubuntu under a QEMU VM.
I will try build a fresh Ubuntu 16.04.2 QEMU image installation to confirm. I remember i did not install any additional package.  ( default + apt-get dist-upgrade only );

dl2ecc, 2017-07-16:
I followed your instructions.
Python 2.7.10 is available now.
I completely installed Opendronemap from ZIP-file new.

Then I started config.sh 
and got these results

---------------------------------------------------------------------------------------------
CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/dl2ecc/OpenDroneMap/build/CMakeFiles/CMakeOutput.log".
See also "/home/dl2ecc/OpenDroneMap/build/CMakeFiles/CMakeError.log".
Configuration Finished

What is missing ?
Thanks for your help
Wolfgang

yjmenezes, 2017-07-16:
@dl2ecc 

I performed a fresh install and all runs fine for me.
---  my steps to install ODM v 0.3.1:
--- 1- fresh Ubuntu 16.4 x64 Desktop
--- 2- as root
sudo su 
apt-get update; apt-get dist-upgrade; apt-get autoremove; exit
--- 3- tools
sudo apt-get install git-core
pip install --upgrade pip
--- 4- as unprivileged user
cd ~/
git clone https://github.com/OpenDroneMap/OpenDroneMap.git
--- 5-
cd ~/OpenDroneMap
./configure.sh  install  2> /tmp/erro.txt
grep -i error /tmp/erro.txt

i  run test too.
please, try git ODM.

dl2ecc, 2017-07-17:
Hi yjmenezes

thanks for your answer.
I'm using ODM for about 2 years in it's first version.
It worked really fine.
Then I tried newer wersion and had no success in installing this versions.

I will start with a blank version of Ubuntu 16.04 an my laptop within this week.
I hope I can run ODM as soon as possible.
Regards
Wolfgang

dakotabenjamin, 2017-07-17:
Yeah maybe something interacting weird with the old version. A fresh install should fix. 

I hope you will be impressed by the improvements since 2 years ago!

pierotofy, 2017-07-18:
Also consider using docker for ease of installation; the performance penalty for using docker is minimal. https://stackoverflow.com/a/26149994

I'm noticing a quite large number of issues related to compilation problems related to ODM of late (not surprisingly, since we have a lot of dependencies). Perhaps we could emphasize the use of docker in the README, as I think most people simply miss the paragraph that states docker is an option.

dl2ecc, 2017-07-20:
Thanks all !!!
I made a completely new installation of Ubuntu 16.04LTS.
Now ODM3 works fine.
The installation process looked very diferent from my first tries.

Best Regards
Wolfgang

dakotabenjamin, 2017-07-20:
Glad to hear make sure you check out the [wiki](https://github.com/OpenDroneMap/OpenDroneMap/wiki) for updated usage instructions.

closed 2017-07-20 by dakotabenjamin