OpenDroneMap/ODM #628: Beyond Hahog features

hy9be, 2017-07-15:
I am new to odm. Not sure if this is a dummy question. I see odm uses HAHOG in feature detection. But OpenSfm actually supports AKAZE, SIFT, SURF as well. What is the reason behind to pick HAHOG?
In OpenSfm, these are configurable through config.yaml file. But i did not see that file in odm project.

Thanks!

CarlosGrohmann, 2017-07-17:
+1 for adding the other feature detection methods. I'd love to test them all.

dakotabenjamin, 2017-07-17:
Like I said in the the gitter chat, this can be accomplished by editing https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/opensfm.py#L65

There are a lot of unused features of OpenSfm; some are not useful to us. I am working on an update to opensfm right now and there will be more config settings opened up when that happens. For now, I will not add different detection methods because that will take a lot of testing, but I will look favorably on pull requests to do so.

hy9be, 2017-07-19:
@dakotabenjamin Hey I am working on a PR to expose some other configs in opensfm. Would you mind sharing me the irrelevant items in the config file: https://github.com/mapillary/OpenSfM/blob/master/config.yaml?

dakotabenjamin, 2017-07-20:
I haven't looked at everything in the sample file, but I know for example SURF and SIFT are both patented so we don't add them to ODM. I don't know about AKAZE, geometric estimation, track creation or navigation graphs. Bundle adjustment is relevant, but I think more for advanced usage. I'm currently working on the GPS param exposure because it affects other parts of the pipeline.

hy9be, 2017-07-20:
AKAZE is included in OpenCV3, so I guess it is patent-free: http://docs.opencv.org/master/d5/d51/group__features2d__main.html. 

I exposed the AKAZE feature type and its parameters. We will need to use the opensfm-0.1.0 rather than the one for odm, as it stucks in the process.

The only thing I am not sure is that, for aerial images, does AKAZE really has the potential to work better than HAHOG. I tried it out yesterday on my case. It was faster, but visually I do not think it delivers a better result.

hy9be, 2017-07-20:
Let me reopen the issue so that may people could see it and may contribute some ideas.

dakotabenjamin, 2017-07-20:
Yes plan to merge opensfm 0.1.0 because that's where I'm working as well.

dakotabenjamin, 2017-08-01:
Hey @hy9be I think that config.yaml file is not well updated, use this: 
https://github.com/mapillary/OpenSfM/blob/8097631c090679c8ed64b083c844b76ab357a0a6/opensfm/config.py

hy9be, 2017-10-25:
I recently worked on a image recognition project using convolutional neural network. The feature extraction ability of CNN is really amazing. Not sure if it will be a good idea to try out using CNN feature instead of Hahog or SWIFT in ODM. What you guys think?

smathermather, 2019-11-14:
@hy9be -- CNNs for feature extraction are an interesting space. Any more thoughts on this?

smathermather, 2020-04-14:
This is implemented.

closed 2020-04-14 by smathermather

sbonaime, 2023-02-03:
> This is implemented.

Are you taking about CNNs ? If yes, how can we try this feature ? 
Thanks