OpenDroneMap/ODM #631: Show the latest README information on opendronemap.org

pierotofy, 2017-07-19:
It looks like the web page at http://opendronemap.org is showing a README from a while ago; it would be beneficial to show users the latest version. cc @smathermather

pierotofy, 2017-07-22:
FYI, to keep it up to date go to Settings:

[image]

Then change the branch to "master" instead of "gh-pages".

[image]

dakotabenjamin, 2017-07-31:
That doesn't actually work, because of the way we were building the website.

pierotofy, 2017-07-31:
After switching to master branch, the URL at https://opendronemap.github.io/OpenDroneMap/ was correctly showing the latest README information. For the domain name to reflect the changes, the "Custom domain" field needed to be set to "opendronemap.org", otherwise github does not know how to associate the two, otherwise you'll get a 404. Hope this makes sense.

dakotabenjamin, 2017-07-31:
Thanks, fixed.

closed 2017-07-31 by dakotabenjamin