OpenDroneMap/ODM #632: Docker GCP Georeferencing failure

mcnis003, 2017-07-19:
I have run ODM successfully several times without GCPs. When GCPs are included the system fails at the georeferencing stage. I have tried running with several  different command configurations. 

`docker run -it -v /path/rgb_autoexposure_calibration_7_4_2_geo/gcp_list.txt:/code/images/gcp_list.txt -v /path/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /path/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /path/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2 --gcp /code/images/gcp_list.txt`

`Finalizing texture atlases... done. (Took: 13.853s)
Building objmodel:
Saving model... done.
Whole texturing procedure took: 327.451s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   /code/images/gcp_list.txt
[INFO]    Found /code/images/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -imagesPath /code/images_resize -imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /code/images/gcp_list.txt -outputCoordFile /code/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
[ERROR]   Georeferencing failed. 
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 144, in process
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Wed Jul 19 05:04:52  2017`

my data:
[https://drive.google.com/open?id=0B8jmw-szgYFGVHYzeTlTSE9PcHc](url)

dakotabenjamin, 2017-07-19:
Have you tried `--skip-resize`?

yjmenezes, 2017-07-19:
@mcnis003 

your images are 1248x950
first try @dakotabenjamin  suggestion  --skip-resize.

or check your  odm_georeference.log.
if your images are resized to 2400x1800, try scale your gcp_list.txt to fit resized images.

mcnis003, 2017-07-19:
Adding --skip-resize did not appear to work. The process appears to have skipped the resizing process, but I received the same error. 

Terminal code:
`docker run -it -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo/gcp_list.txt:/code/images/gcp_list.txt -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2 --gcp /code/images/gcp_list.txt --skip-resize`

I believe the resize step was skipped; these are the original sizes of my photos:
`[INFO]    Found 103 usable images
[INFO]    Running ODM Load Dataset Cell - Finished
[INFO]    Running ODM Resize Cell
[DEBUG]   Resizing dataset to: /code/images_resize
[DEBUG]   Copied calibrated_0cae8e93088d-FW00067.jpg | dimensions: (950, 1248, 3)
[DEBUG]   Copied calibrated_0cae8e93088d-FW00053.jpg | dimensions: (950, 1248, 3)
[DEBUG]   Copied calibrated_0cae8e93088d-FW00068.jpg | dimensions: (950, 1248, 3)
[DEBUG]   Copied calibrated_0cae8e93088d-FW00054.jpg | dimensions: (950, 1248, 3)
[DEBUG]   Copied calibrated_0cae8e93088d-FW00069.jpg | dimensions: (950, 1248, 3)`

Error:
`Building objmodel:
	Saving model... done.
Whole texturing procedure took: 278.924s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   /code/images/gcp_list.txt
[INFO]    Found /code/images/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -imagesPath /code/images_resize -imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /code/images/gcp_list.txt -outputCoordFile /code/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
[ERROR]   Georeferencing failed. 
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 144, in process
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Wed Jul 19 21:55:12  2017`

I will try rescaling my GCP coordinates, run again, and check the log file for the actual sizes of my photos.

yjmenezes, 2017-07-19:
@mcnis003 

1- first  try without gcp:     --use-exif
2- replace the gcp_list.txt
from: WGS84	UTM	15N
to:  EPSG:32615

mcnis003, 2017-07-19:
> first try without gcp: --use-exif
>check the directory resized_images/

I ran without GCP and without --skip-resize:

Terminal command:
`caffeinate docker run -it -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2`

Resized image info:
`file calibrated_0cae8e93088d-FW00201.jpg calibrated_0cae8e93088d-FW00201.jpg: JPEG image data, JFIF standard 1.01, aspect ratio, density 1x1, segment length 16, Exif Standard: [TIFF image data, little-endian, direntries=10, manufacturer=Sentera, model=RGB_1.2MP-GS-0002_6.00mm-0001_0000, xresolution=178, yresolution=186, resolutionunit=1, software=2.1.26, datetime=2017:07:03 16:44:50, GPS-Data], baseline, precision 8, 2400x1827, frames 3
root@95ff47313e4a:/code/images_resize# `

The images are being resized.

I then ran without GCP and with --skip-resize:
`docker run -it -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo/gcp_list.txt:/code/images/gcp_list.txt -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2 --gcp /code/images/gcp_list.txt --skip-resize`

Resized image info:
`calibrated_0cae8e93088d-FW00200.jpg
calibrated_0cae8e93088d-FW00127.jpg  calibrated_0cae8e93088d-FW00201.jpg
calibrated_0cae8e93088d-FW00128.jpg  odm_textured_model_material0000_map_Kd.png
calibrated_0cae8e93088d-FW00129.jpg  odm_textured_model_material0001_map_Kd.png
calibrated_0cae8e93088d-FW00130.jpg  odm_textured_model_material0002_map_Kd.png
root@3a739a1d6015:/code/images_resize# file calibrated_0cae8e93088d-FW00200.jpg calibrated_0cae8e93088d-FW00200.jpg: JPEG image data, JFIF standard 1.01, aspect ratio, density 96x96, segment length 16, Exif Standard: [TIFF image data, little-endian, direntries=10, manufacturer=Sentera, model=RGB_1.2MP-GS-0002_6.00mm-0001_0000, xresolution=178, yresolution=186, resolutionunit=1, software=2.1.26, datetime=2017:07:03 16:44:49, GPS-Data], baseline, precision 8, 1248x950, frames 3
root@3a739a1d6015:/code/images_resize# `

The system successfully skipped the resize step. I don't think I need to allow this process to complete right now because I have already successfully ran ODM on these images several times without GCP. I can do this overnight help troubleshoot if needed.

I will try this next and report when it completes:
> 2- replace the gcp_list.txt
> from: WGS84	UTM	15N
> to: EPSG:32615

mcnis003, 2017-07-20:
Ran with --skip-resize and with modified GCP file as suggested by @yjmenezes 

`docker run -it -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo/gcp_list.txt:/code/images/gcp_list.txt -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2 --gcp /code/images/gcp_list.txt --skip-resize`

Result, failure again:

`INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   /code/images/gcp_list.txt
[INFO]    Found /code/images/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -imagesPath /code/images_resize -imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /code/images/gcp_list.txt -outputCoordFile /code/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
[ERROR]   Georeferencing failed. 
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 144, in process
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Thu Jul 20 01:33:58  2017`
 
Is it possible that the failure is caused by the low precision of my GCPs? The photos are taken at about 1inch/pixel but I only have a handheld GPS unit that records to the tenth of a second Lat and Long. I know that is about 8 feet.

mcnis003, 2017-07-20:
Successful completion without GCPs:

`docker run -it -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/images -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_orthophoto -v /Users/ianmcnish/Desktop/rgb_autoexposure_calibration_7_4_2_geo:/code/odm_texturing my_odm_image --orthophoto-resolution 30.0 --opensfm-processes 2`

`Building objmodel:
	Saving model... done.
Whole texturing procedure took: 381.248s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /code/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -inputCoordFile /code/odm_georeferencing/coords.txt -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   Line: WGS84 UTM 15N`

...

`[DEBUG]   lat_frac: 4981819/1 38/1 33180/1000
[DEBUG]   running /code/SuperBuild/build/pdal/bin/pdal pipeline -i /code/odm_georeferencing/las.json --readers.ply.filename=/code/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/code/odm_georeferencing/odm_georeferenced_model.las
[INFO]    Creating geo-referenced CSV file (XYZ format)
[WARNING] Found a valid georeferenced model in: /code/odm_georeferencing/odm_georeferenced_model.ply
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running ODM DEM Cell
[DEBUG]   running l2d_classify --help > /dev/null
[INFO]    Create DSM: False
[INFO]    Create DTM: False
[INFO]    DEM input file /code/odm_georeferencing/odm_georeferenced_model.las found: True
[WARNING] DEM will not be generated
[INFO]    Running ODM DEM Cell - Finished
[INFO]    Running ODM Orthophoto Cell
[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /code/odm_texturing/odm_textured_model_geo.obj -logFile /code/odm_orthophoto/odm_orthophoto_log.txt -outputFile /code/odm_orthophoto/odm_orthophoto.png -resolution 30.0  -outputCornerFile /code/odm_orthophoto/odm_orthophoto_corners.txt
[DEBUG]   Line: WGS84 UTM 15N

[INFO]    Creating GeoTIFF
[DEBUG]   running gdal_translate -a_ullr 485288.047222 4981867.91476 485412.874454 4981730.5386 -co TILED=yes  -co BIGTIFF=IF_SAFER -co COMPRESS=DEFLATE -co PREDICTOR=2  -co BLOCKXSIZE=512 -co BLOCKYSIZE=512 -co NUM_THREADS=ALL_CPUS -a_srs "EPSG:32615" /code/odm_orthophoto/odm_orthophoto.png /code/odm_orthophoto/odm_orthophoto.tif > /code/odm_orthophoto/gdal_translate_log.txt
[INFO]    Running ODM OrthoPhoto Cell - Finished
[INFO]    OpenDroneMap app finished - Thu Jul 20 03:59:46  2017
`

mcnis003, 2017-07-20:
I ran again with much more precise coordinates and that failed too.

yjmenezes, 2017-07-20:
@mcnis003 

please,
1- check the image size inside /code/images_resize 
2- compress and share the new gcp_list.txt and /code/odm_georeferencing/odm_georeferencing_log.txt 
3- try check the gcp over the --use-exif output   /code/odm_orthophoto/odm_orthophoto.tif  ( under QGIS for example )

mcnis003, 2017-07-20:
> 1- check the image size inside /code/images_resize

> 2- compress and share the new gcp_list.txt and /code/odm_georeferencing/odm_georeferencing_log.txt

Here is a link to a folder with the new gcp_list:
https://drive.google.com/open?id=0B8jmw-szgYFGQjBqQzFLaG9xZzg

@yjmenezes I am not positive how to get the odd_georeferencing_log.txt for the stopped container. Is there someway to output this before the process completes?

> I actually produced the new GCPs using the output of the --use-exif odm_orthophoto.tif. The points appear to be accurate. You can find those results here:
https://drive.google.com/open?id=0B8jmw-szgYFGaWg2ZWFLNEhMU2M

mcnis003, 2017-07-20:
I notice that the console out contains this during the georeferencing step:
`-bundleResizedTo 2400`

My photos are definitely not being resized when I use the --skip-resize option.

dakotabenjamin, 2017-07-20:
@mcnis003 that is very clearly a bug! Thanks for pointing that out.

mcnis003, 2017-07-20:
Here is the console data:

`Whole texturing procedure took: 246.866s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   /code/images/gcp_list.txt
[INFO]    Found /code/images/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -imagesPath /code/images_resize -imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /code/images/gcp_list.txt -outputCoordFile /code/odm_georeferencing/coords.txt
Error in Georef:
Fewer than 3 GCPs have correspondences in the generated model.
[ERROR]   Georeferencing failed. 
Traceback (most recent call last):
  File "/code/scripts/odm_georeferencing.py", line 144, in process
    '-outputCoordFile {coords}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
[INFO]    OpenDroneMap app finished - Thu Jul 20 06:28:12  2017`

yjmenezes, 2017-07-20:
@mcnis003 

attached your new gcp_list.txt, fixed a double tab.
this gcp are measured over your raw images ( 1248wx950h)

please, check the image size inside  /code/images_resize
check if it has been resized to 2400.
if resized, your gcp list must be scaled or measured over resized images.

btw: --use-exif take positions from photos exiftags ( navegation lat/lon +- 2.5 m or more );

[gcp_list.txt.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1162540/gcp_list.txt.zip)

mcnis003, 2017-07-20:
@yjmenezes I have a question about rescaling my gcp coordinates. The original files are list by ODM as 950, 1248 (height, width) and the resized images are list as 2400, 1827 (width?, height?).

yjmenezes, 2017-07-20:
@mcnis003 
good question, i guess scale factor will be 2400/1248.
please, try attached gcp_list_scaled.txt
[gcp_list_scaled.zip](https://github.com/OpenDroneMap/OpenDroneMap/files/1162596/gcp_list_scaled.zip)

mcnis003, 2017-07-20:
@yjmenezes I will run that, thanks. Also, here are the data from images_resized without --skip-resize.

`file calibrated_0cae8e93088d-FW00200.jpg 
calibrated_0cae8e93088d-FW00200.jpg: JPEG image data, JFIF standard 1.01, aspect ratio, density 1x1, segment length 16, Exif Standard: [TIFF image data, little-endian, direntries=10, manufacturer=Sentera, model=RGB_1.2MP-GS-0002_6.00mm-0001_0000, xresolution=178, yresolution=186, resolutionunit=1, software=2.1.26, datetime=2017:07:03 16:44:49, GPS-Data], baseline, precision 8, 2400x1827, frames 3
root@f12a386c5c1d:/code/images_resize# `

mcnis003, 2017-07-20:
Success!

`Whole texturing procedure took: 344.133s
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   /code/images/gcp_list.txt
[INFO]    Found /code/images/gcp_list.txt
[DEBUG]   running /code/build/bin/odm_georef -bundleFile /code/opensfm/bundle_r000.out -imagesPath /code/images_resize -imagesListPath /code/opensfm/list_r000.out -bundleResizedTo 2400 -inputFile /code/odm_texturing/odm_textured_model.obj -outputFile /code/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /code/opensfm/depthmaps/merged.ply -outputPointCloudFile /code/odm_georeferencing/odm_georeferenced_model.ply  -logFile /code/odm_georeferencing/odm_georeferencing_log.txt -outputTransformFile /code/odm_georeferencing/odm_georeferencing_transform.txt -georefFileOutputPath /code/odm_georeferencing/odm_georeferencing_model_geo.txt -gcpFile /code/images/gcp_list.txt -outputCoordFile /code/odm_georeferencing/coords.txt
[DEBUG]   Line: EPSG:32615

[DEBUG]   lat_frac: 4981825/1 35/1 45240/1000
[DEBUG]   lat_frac: 4981822/1 39/1 18000/1000
[DEBUG]   lat_frac: 4981816/1 9/1 25776/1000
[DEBUG]   lat_frac: 4981810/1 13/1 16536/1000`

Rescaling the GCP and omiting --skip-resize works.  I would guess that it has something to do with the `-bundleResizedTo 2400` flag in the georeferencing step.

I will run another dataset and comment on alignment later today.

> btw: --use-exif take positions from photos exiftags ( navegation lat/lon +- 2.5 m or more );
Thanks for the info. Alignment between photos in more important than alignment of the photos to actual position.

closed 2017-07-20 by mcnis003

dakotabenjamin, 2017-07-21:
Yeah it seems that --skip-resize is bugged for odm_georeferencing