OpenDroneMap/ODM #633: Relationship between orthophoto pixel coordinates and and source image(s)

piinecone, 2017-07-22:
Is it possible to determine the filenames of the input images that correspond to a given (x,y) on the resultant orthophoto?

For example, for a given point (x,y) or rect, I would like to know which input images contributed to that portion of the orthophoto. The end goal is to see the preprocessed image or images that correspond with a specific area of the orthophoto.

This is more of a clarification or feature request than an issue, so I'm happy to move this to stackoverflow if that's better.

yjmenezes, 2017-07-22:
@piinecone 

i wrote shell scripts  to take photo location from exiftags and export a TXT with photo name plus   lat/lon or UTM.
I use it to select the photos  i wanna process, plotting with QGIS..
It is not exactly what you are looking for but maybe it helps you.

https://github.com/yjmenezes/

piinecone, 2017-07-22:
Hi @yjmenezes, thanks for your response and for sharing your scripts! Unfortunately they're not quite what I'm looking for.

I've decided to try to understand how input photos lead to textures, because I suspect that the `renderPixel` function is the right place to map pixel coordinates to textures. A second map of textures to input image filenames would allow me to solve the lookup from (x,y) to input image.

I'll post any progress in this thread.

yjmenezes, 2017-07-22:
@piinecone 

it is interesting ! 
Sometimes we have bad floating points projected over orthophoto. 
[image]

piinecone, 2017-07-24:
I have a brief update on this. I don't have a pixel perfect mapping at the moment, but I'm able to extract the center of the orthophoto (in UTM coordinates) from the `odm_georeferencing/odm_georeferencing_model_geo.txt` and the corners from `odm_orthophoto/odm_orthophoto_corners.txt` to build a UTM rect of the orthophoto. I can then find the corresponding UTM coordinates for each (x,y) of the image, and look up the nearest images from the input set.

Next, I will try to determine a best image (perhaps using a sharpness score or, better, some output from ODM) and see if applying the rotations and translations from `reconstruction.json` helps orient the source image to its role in the orthophoto.

yjmenezes, 2017-07-26:
@piinecone 

here are the bad floating textures efects i was talking about.
your idea can help to fix it, i guess.

[image]

dakotabenjamin, 2017-07-26:
@yjmenezes THe floating texture effects should be better handled by using `--use-25d-mesh` have you tried that? 

@piinecone I don't think we have what you are looking for but I will link to an issue I opened over at mvs-texturing: https://github.com/nmoehrle/mvs-texturing/issues/75 if that adds any insight

yjmenezes, 2017-07-26:
@dakotabenjamin,

 thx,  your  guidance is highly appreciated, i will try --use-25dmesh.

outlier  removal and DSM filter to get  DEM are challenges for me. We have a large set of ODM options to learn. We go ahead;

yjmenezes, 2017-07-26:
@dakotabenjamin 

thank you,  using --use-25 mesh, orthophoto seems better, just the  small area has some noise.
[image]

dakotabenjamin, 2017-07-26:
That's good. Maybe this will help as well: 
```
--mesh-wlop-iterations <positive integer>
                        Iterations of the Weighted Locally Optimal Projection
                        (WLOP) simplification algorithm. Higher values take
                        longer but produce a smoother mesh. Applies to 2.5D
                        mesh only. Default: 35
```

yjmenezes, 2017-07-26:
@dakotabenjamin 

--mesh-wlop-iterations  35 , new noises. I will test with other data sets.
thank you.

[image]

dakotabenjamin, 2017-07-26:
Did you try increasing? 35 is the same as the default. 

More to the original issue, it may help to implement a point cloud outlier removal. I think some of this already happens but maybe we need to increase, or be able to manipulate the level of removal. 

The other thing to implement is producing a model of the seam-lines. That could be done as a shape file I think.

ArturoBurela, 2017-12-02:
@piinecone 
Im developing some tools using ODM. I write you to aske for a little help.

You mentioned this:

> I have a brief update on this. I don't have a pixel perfect mapping at the moment, but I'm able to extract the center of the orthophoto (in UTM coordinates) from the [...]

I understand the UTM coordinates in _model_geo.txt, for example:
WGS84 UTM 17N
437036 4572789

But in _orthophoto_corners i have: -2.06128997802734375e+02 -1.53876937866210938e+02 1.80876098632812500e+02 1.72307052612304688e+02

I dont understand what they mean.
Could you please give help me?? I need to know the corresponding UTM for each X,Y as you mentioned.

smathermather, 2018-11-22:
Fixed with https://github.com/OpenDroneMap/ODM/commit/ee5353350c5df449e69242da9b049cbef25643e7.

Well, earlier really, but definitely fixed now. Closing.

closed 2018-11-22 by smathermather