OpenDroneMap/ODM #634: Fresh install native Ubuntu16.04 error

KommandorKeen, 2017-07-24:
Just did a native install on Ubuntu 16.04 so that I can access the code and contribute, but...

  File "run.py", line 8, in <module>
    import ecto
  File "/home/simon/2TB/odm/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

This is a noob problem so any kind pointers would be helpful.

starsep, 2017-07-24:
Look at https://github.com/OpenDroneMap/OpenDroneMap#installation

Also issue is duplicate of #420

KommandorKeen, 2017-07-24:
Thanks,
Updated my ~/.bashrc but forgot to logout to re-read it. 
Solved the problem by running the /my/path/OpenDroneMap/run.sh
Thanks for your patience.

closed 2017-07-24 by KommandorKeen