OpenDroneMap/ODM #636: How to combine textures?

flyandi, 2017-07-25:
I have successfully processed a lot of different stitching projects however I fail to understand how to combine all the texture files in one big happy file. 

Does ODM has a command line switch that I may missed? Or is this usually done through intensive post processing, and if so, how?

Any guidance is highly appreciated.

dakotabenjamin, 2017-07-26:
It is not something ODM is capable of yet. We are working to some scaling improvements by the end of the year which will be able to split large datasets and then merge them, which will include textured models as well. But for now I think manual post-processing is necessary. A quick google didn't find much but this video seems to describe what you want at around the 15 min mark: https://vimeo.com/14783202

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/how-to-combine-textures/368

closed 2018-03-28 by dakotabenjamin