OpenDroneMap/ODM #637: Open Drone Map issue while computing neighbours, KeyError: 'plane is not a file in the archive'

aaronstone007, 2017-07-26:
Error as follows :

Traceback (most recent call last):
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
    dense.compute_depthmaps(data, graph, reconstructions[0])
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 38, in compute_depthmaps
    parallel_run(clean_depthmap, arguments, processes)
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 47, in parallel_run
    return [function(arg) for arg in arguments]
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 109, in clean_depthmap
    add_views_to_depth_cleaner(data, reconstruction, neighbors[shot.id], dc)
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 189, in add_views_to_depth_cleaner
    depth, plane, score = data.load_raw_depthmap(shot.id)
  File "/home/ubuntu/ODM/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 100, in load_raw_depthmap
    return o['depth'], o['plane'], o['score']
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 228, in __getitem__
    raise KeyError("%s is not a file in the archive" % key)
KeyError: 'plane is not a file in the archive'
Traceback (most recent call last):
  File "run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/ubuntu/ODM/OpenDroneMap/scripts/opensfm.py", line 128, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/ubuntu/ODM/OpenDroneMap/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

dakotabenjamin, 2017-07-27:
Can you describe your issue more? I need more information to diagnose the problem.

aaronstone007, 2017-07-28:
Hello @dakotabenjamin it was a problem with the images, i re-ran with new set of images it worked fine,but the documentation is a little off at http://opendronemap.org/ in configure.sh line

closed 2017-07-28 by aaronstone007

dakotabenjamin, 2017-07-31:
Sorry, we have to update the website. Thanks for reporting!