OpenDroneMap/ODM #638: [Docker] results volumes permissions

merkato, 2017-07-26:
After processing dataset with docker container, I've noticed that resulting files and directories are created with root owner. How to change this behavior?

command line
```
docker run -it --rm \
-v $(pwd)/images:/code/images \
-v $(pwd)/odm_orthophoto:/code/odm_orthophoto \
-v $(pwd)/odm_texturing:/code/odm_texturing \
-v $(pwd)/odm_dem:/code/odm_dem \
-v $(pwd)/odm_georeferencing:/code/odm_georeferencing \
opendronemap/opendronemap --dsm --dtm```

ls result
```
drwxrwxr-x 2 1000 1000 4096 lip 25 19:56 images
drwxr-xr-x 2    0    0 4096 lip 26 01:32 odm_dem
drwxr-xr-x 2    0    0 4096 lip 25 22:59 odm_georeferencing
drwxr-xr-x 2    0    0 4096 lip 26 01:33 odm_orthophoto
drwxr-xr-x 2    0    0 4096 lip 25 22:52 odm_texturing
-rwxrwxr-x 1 1000 1000  276 lip 25 19:56 run.sh
```

dakotabenjamin, 2017-07-26:
I run into this as well. I'm not too keen on how docker works but maybe this will help?

https://stackoverflow.com/questions/40643169/docker-run-with-volume-changes-directory-permission-to-root

pierotofy, 2017-07-26:
You could try passing `-u `id -u $USER`` to the command (right after --rm) to execute the command under a certain userID. This might or might not work, but it's worth a try.

The problem is that users in the docker environment are not directly mapped to users in the host environment (neither should they be). If could be possible to perhaps add a patch to ODM that sets the permissions to the output folders based on the userid/groupid of the input images, that would probably fix it.

merkato, 2017-07-26:
This doesn't work on my machine (Xubuntu 17.04), with latest odm image.

yjmenezes, 2018-03-18:
@merkato 

try  this, please

#to avoid sudo when running docker

sudo gpasswd -a $USER docker
source ~/.bashrc

dakotabenjamin, 2018-03-28:
Docker builds have changed since this issue opened. If this is still a problem, please reopen.

closed 2018-03-28 by dakotabenjamin