OpenDroneMap/ODM #641: hosting a ODM discourse forum?

PeterSprague, 2017-07-29:
I am willing to host an ODM discourse forum to increase support and create a living documentation for users.  Would the community use this to build out faqs?

I know there is a gitter help section, but dislike gitter for any historical referencing and building knowledge documentation.  Gitter seems to be for immediate chat, not building a depth of documentation.

pierotofy, 2017-07-29:
A migration to discourse has actually been in the talks for a few months now; Cleveland Metroparks has already offered to host it and I think they are getting ready to set it up. cc @smathermather any ETA on this one?

PeterSprague, 2017-07-29:
I could have it up in a docker container this weekend, but will hold off until hear further.

Glad to see it has been discussed, wondering what the holdup is?

dakotabenjamin, 2017-07-31:
We got an internal server up, I just don't have access yet. This week we should have a preview version up.

dakotabenjamin, 2018-03-28:
THis is up now! http://community.opendronemap.org

closed 2018-03-28 by dakotabenjamin