OpenDroneMap/ODM #643: Output mangled or application crash when having duplicates in dataset

fredlllll, 2017-07-31:
reproduce issue:
get one of the demo datasets and duplicate ~30 images into the same folder.
process dataset

result:
either mangled output or even crash while processing

expected:
duplicated datapoints shouldnt add any information to output, much less mangle it

rudimentary solution: scan for identical files before processing and only use one of them

another problem with this: images that are taken from almost the same place. i have only had one in our dataset where the problem arose, but i guess that would cause similar problems to just duplicate files

dakotabenjamin, 2017-07-31:
What kind of errors are you getting?

fredlllll, 2017-07-31:
i didnt record it =/ we built an automated system around opendronemap to process datasets from drones, and when processing the aukerman with 30 duplicated images, it just returned an empty error (usually that should contain the whole odm output), so i cant tell. i will run it again locally tomorrow if i can (my pc isnt the fastest and work also uses some cpu)

mcnis003, 2017-08-06:
I get the same thing. I backup my images on Google Drive. Sometimes this causes duplicate files to be created. I had to write some code to remove these before running ODM. I don't have a copy of the error text but I believe the crash was during calculating depth maps or merging depth maps. I can recreate it if needed.

fredlllll, 2017-08-08:
@mcnis003 i would appreciate it, as my computing capabilities are rather limited =/

mcnis003, 2017-08-08:
I did it R because that is what I am most familiar with:

`list=list.files(path="/path/to/files",pattern="*pattern_here*.jpg",recursive = TRUE,full.names = TRUE)
for (i in 1:length(list)){
  if (file.exists(list[i])) file.remove(list[i])
}`

It could also be implemented in Bash using grep and rm if you are on Linux or Mac. Basically:

1. create a list of files that are duplicates using some sort of pattern recognition, in my case the duplicates are file_name(1).jpg so I search for the "(1)".
2. delete those files

fredlllll, 2017-08-08:
oh i thought you were talking about the error text

pierotofy, 2020-09-11:
ODM is capable of handling duplicate images well in the latest version (just confirmed with a duplicate brighton dataset):

[image]

:+1:

closed 2020-09-11 by pierotofy