OpenDroneMap/ODM #646: How are those odm_textured_model_material*****_map_kd.pngs generated   ?

hs5530hs, 2017-08-02:
I studied the code in the odm_texturing part, I found the picture output result should be **texture_*.png**  and the patches in the picture are arranged in rectangle, but I found the real picture output result in ubuntu are **odm_textured_model_material*****_map_kd.png** and the patches arrangement are irregular. I checked the  code in others part but still can't find the reasons, can anybody explain it to me? Thanks!

dakotabenjamin, 2017-08-02:
we pull this code in: https://github.com/nmoehrle/mvs-texturing have you looked at that?

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/how-are-those-odm-textured-model-material-images-generated/379

closed 2018-03-28 by dakotabenjamin