OpenDroneMap/ODM #647: MemoryError: out of memory on Z820 with 64GB Ram

PeterSprague, 2017-08-03:
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
    dense.compute_depthmaps(data, graph, reconstructions[0])
  File "/code/SuperBuild/src/opensfm/opensfm/dense.py", line 38, in compute_depthmaps
    parallel_run(clean_depthmap, arguments, processes)
  File "/code/SuperBuild/src/opensfm/opensfm/dense.py", line 50, in parallel_run
    return p.map(function, arguments)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 567, in get
    raise self._value
MemoryError: out of memory
2017-08-03 17:59:35,626 Cleaning depthmap for image 2017_0713_152053_070.jpg
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/opensfm.py", line 128, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1

dakotabenjamin, 2017-08-03:
How many images? and did you change `--opensfm-processes`? This is a known bug and the current workaround is to reduce the number of processes until we get those changes worked in (sometime late September I hope).

PeterSprague, 2017-08-03:
Same 150 images.  Yup, set opensfm-processes explicitly to my nproc of 32.  Rerunning now with default.

Why the difference when opensfm-processes defaults to $(nproc) accordingto the settings.yaml directions?

dakotabenjamin, 2017-08-03:
sorry if I misunderstand, you have 32 cores and explicitly set to 32? That should be no different from the default nproc. I would suggest 16, it will take longer but won't run into memory errors.

PeterSprague, 2017-08-03:
Yup, wanted to test explicitly.  Seemed to be subjectively faster, though had changed a few other parameters.  Reran with default opensfm-processes, cpus were just idling.  Set to 16, now cpu are showing use

PeterSprague, 2017-08-03:
At the merging depthmaps stage, memory usage kept climbing to 100% Ram (62.8gb) + 100% swap space (22.4gb) with opensfm-processes: 16 (16/32 cores)  Didn't error out this time, and memory usage dropped quickly once past merging stage.

Runs on same data were taking up to five hours with default, this one was ~1.5hrs

dakotabenjamin, 2017-08-04:
IT sounds like you worked out your issue, so I'll close. Feel free to reopen if other problems arise.

PeterSprague, 2017-08-04:
I did by lowering the explicit cores specified to 12 out of 32, but there seems to be some discrepancy on what is default.  In settings.yaml from master says open-sfm-processes defaults to nproc, which would be 32 in my case.  In wiki says opensfm-processes: 4 is default

If run default hardly any cores are actually called, which indicates that the code and/or notes in yaml file are not as specified.  To get any performance, one needs to explicitly specify the cores called by opensfm.

I would still consider this an open issue for the project.  Any pointers in the code where I should start looking beyond the trace from above?

dakotabenjamin, 2017-08-08:
I updated the wiki. which can sometimes be outdated.

Here's all the instances where the number of processes are brought up in OpenSfM: https://github.com/mapillary/OpenSfM/search?utf8=%E2%9C%93&q=+data.config.get%28%27processes%27%2C+1%29&type=

The number passed from ODM is set either inline `--opensfm-processes` or in the config file. That number is used to programmatically write the config.yaml file for opensfm found in the opensfm data directory. That line of code is here: https://github.com/OpenDroneMap/OpenDroneMap/blob/master/scripts/opensfm.py#L69

dakotabenjamin, 2017-08-08:
For clarity, I think we set opensfm-processes to 1 by default because of some memory issues that are being worked out. That may also be a source of confusion.

KommandorKeen, 2017-08-08:
@dakotabenjamin That was the same thing that confused me when it was changed. The default IS one core, but I try to run on 3 or 4 with 40GB of RAM and 200 images.  If I set the number of features too high I rapidly run out of RAM.

xialang2012, 2017-08-17:
Using the newest version of the OpenSfM to replace the one which used by the ODM, the problem will be solved.

dakotabenjamin, 2017-08-24:
#644 Updates opensfm to bring in some optimization improvements but OpenSfM version 0.1.0 doesn't have the latest changes. Waiting on those to be released in 0.2.0. Please test the new changes if you have a chance!

KommandorKeen, 2017-08-28:
dakotabenjamin
I have stock version o.3.1 running natively on my Ubuntu 16.04 
what is the least painful path to updating to include #644?
Thanks

dakotabenjamin, 2017-08-28:
You will have to clone the git repository in a new folder:
```
git clone https://github.com/OpenDroneMap/OpenDroneMap
git checkout -b dakotabenjamin-update-vtk master
git pull https://github.com/dakotabenjamin/OpenDroneMap.git update-vtk
./install.sh install
./run.sh ...
```

dakotabenjamin, 2017-09-01:
#644 has been merged so you can pull from master now.

closed 2017-09-01 by dakotabenjamin

PeterSprague, 2017-09-24:
Did some test runs yesterday.  Seems to be subjectively faster to build by ~20-30%, but opensfm parameter does not seem to have any effect.  Used opensfm not specified aka default, opensfm=10 (was best could use with July Docker build), opensfm=32 on same set of 164 images.  All runs took ~01:15

Does the opensfm parameter setting in settings.yaml control the cores used, or is it now hard-coded in?

dakotabenjamin, 2017-09-26:
settings.yaml will control the number of processes used, it's not hardcoded. It you're using docker, you have to use the command line or edit docker.settings.yaml and rebuild the image (which would be inconvenient).