OpenDroneMap/ODM #648: Point cloud density and smoothing/de-noising for more uniform "surfaces" discussion

rttgnck, 2017-08-04:
I am new to this project and don't think I can offer much in terms of help. I would however like to open discussion on a few things though so as to better understand the project.

I have noticed that even when processing a small close dataset, 6 photos from 30-50ft away that the density of the point cloud is very small. Almost like the density of the point cloud scaled with the size of the overall image. Is there a limiting variable for the point cloud's density (XX px/cm?) or is it currently a limitation of the software that it can only produce/process so many point at once?

The other part would be how the point cloud is generated and how surfaces that should be flat are not. I am wondering how exactly the surfaces are calculated (surfaces being walls of shares point like on the face of the building). Would there be a place to look at tightening the surfaces irregularity. Right now on a close surface, a brick face, the point vary by as much as a few inches in reality. I assume with an increase in point cloud density then the ability to average nearest points would be able to be tightened and thus improve surface smoothness. Would there be a way to choose outliers that would otherwise be thrown away, or to far from the nearest average and compute an offset for them so you can pull them back in to the local average and in turn smooth the surface.

Please forgive my ignorance. I am really interested in the software and it's abilities.

I would like to see this scalable and distributable so that a site can be processed in chunks across multiple nodes (multiple machines) in parallel and then joined back together. This would really only speed up processing time for users with access to multiple machines and it would only be useful if the point cloud could be increased in density so that the smaller chunks can have as many point as the whole site, and thus increase the overall density of the site. Once the overall site has been conjoined a secondary scan could be implemented to "correct" outliers outside of a specific range to basically add or subtract an offset in any number of directions and bring them closer together. This could be done by comparing a points color value to surrounding points and then using like colored points to help compute the offset. Ideally it would start with the point closest to the average to correct their locations first and then move further from the average. At that point anything outside a specified standard deviation could be discarded.  Again this falls back to a density increase so there are more points for processing.

KommandorKeen, 2017-08-07:
You can increase the number of points by increasing the number of features detected in the images before pairing. Be careful here as 15,000 points per image x 166 images chews up over 40GB of memory. congrats on getting a workable solution from six images. Have a look at https://smathermather.com/category/3d/opendronemap-3d/ to see where the project might go and I think the blog addresses your thoughts on its future.

rttgnck, 2017-08-07:
I've started looking at OpenSFM and OpenMVS for processing, for now, so I can better understand what's happening. I had glanced through that blog and it's probably the seed for my post. I will be looking at it further.

 Is there another location that here and gitter than may be a bit more where I can follow the work as it progresses?

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/point-cloud-density-and-smoothing-de-noising-for-more-uniform-surfaces-discussion/372

closed 2018-03-28 by dakotabenjamin