OpenDroneMap/ODM #649: SLAM reconstruction problem with latest release: build/bin/odm_slam: not found

DelphiR, 2017-08-04:
I downloaded the latest release and build it in 14 and 16 Ubuntu.
In both cases after all builds with SLAM support and attempt to run the reconstruction I get the same error: build/bin/odm_slam: not found

dakotabenjamin, 2017-08-08:
I haven't looked into the SLAM module in a while, I'm sorry it isn't working for you. I've labeled it as a bug, we will look into it

DelphiR, 2017-08-11:
Thank you. Wait for a news. It is very interesting feature of OpenDroneMap. I can support you bug fix. I can test. Please could be possbile to make a path to exisitng release (3.1) ?

Ntweat, 2017-09-28:
I have solved this problem
Build you do cmake in build directory edit the CMakeCache.txt, fine ODM_BUILD_SLAM:BOOL to ON
and then type make
will compile

smathermather, 2020-04-14:
Video processing has been removed.

closed 2020-04-14 by smathermather