OpenDroneMap/ODM #650: issues with shadows during orthorectification?

mcnis003, 2017-08-06:
I think I am having an issue with shadows during orthorectification.

My setup is a multispectral drone that captures RGB, 655nm, 725nm, and 800nm images with 4 different cameras for agricultural research. I produce 4 separate orthophotos and align using GCPs. I have used ODM for many day's worth of data and I notice the all my photos align perfectly except the 655nm photos. The 655nm orthophotos always have a "bowing" while the other photos are perfect. I am not very familiar with the orthorectification process, but i wonder if this could be due to the shadows in my photos.

I am located in Minnesota and even at solar noon in the summer the sun is in the Southern sky. This means that my photos nearly always have a small shadow to the North. I believe this is causing a problem with the 655nm photos because there is little difference in Dn values between the shadows and the plant since plants absorb most 655nm light. Am I thinking correctly about this problem? Is there a possible fix?

655nm (with bowing):

[image]

RGB (perfect):

[image]

KommandorKeen, 2017-08-07:
Looks more like a timing issue than a shadow issue or a it could be lens calibration distortion issue. Can you share the 655nm data set? 
Simon

mcnis003, 2017-08-07:
here it is:
[https://drive.google.com/drive/folders/0B_c8yadvyVBlbkRLLUMtenMyWmM?usp=sharing](url)

mcnis003, 2017-08-08:
My drone produces a file that records the gps location every couple milliseconds. Using this data I changed the GPS location in the image exif data. Unfortunately this has not fixed the problem.

[image]

The other photos processed perfectly like they always have. Only the 655nm photo had a problem.

KommandorKeen, 2017-08-08:
Agisoft makes the data bow out in the middle and ODM makes it bow in in the  middle. This is pointing to a lens distortion issue. Not much more to say at the moment.

KommandorKeen, 2017-08-08:
OR there is a timing issue with this camera. Will look at this further....

mcnis003, 2017-08-08:
I tried processing just the first half of the data. There are multiple trials in my photos o processing this way is not too much of a burden. I notice that this processes better when compared to a normal RGB photo from another flight.

[image]

[image]

Is this consistent with a lens distortion of timing problem?

KommandorKeen, 2017-08-10:
Not sure exactly how it all comes together in ODM, but I note that your two survey lines at the ends were in the same direction (North) and that the bow in the data is to the North at both ends. This points me towards a timing problem, but it is more gut feel than anything. Drop just one survey line from one end of the dataset and see if the result is straight, but skewed north at 1 end and south at the other. That would nail it a s a timing problem.

mcnis003, 2017-08-12:
I have solved this problem. I believe this is cause by distortion on the right side of my 655_nm images. If you look at the images, objects on the right side of the image appear to be out of focus:

object on left side of image:

[image]

same object on right side of image:

[image]

This problem become progressively worse as the objects move to the right side of the photo. This problem does not appear in the photos for my other cameras.

I attempted to resolve this issue by cropping my photos before ODM. In my case I cropped from 1248x950 to 1000x950. I also had to change the rescaling factor for the GCPs for the 655nm photos, but I did not need to remove GCPs that fall out of the frame of the cropped photo. The result:

[image]

This appears to be a problem with my camera and not an issue with ODM. I will change the subject of the title to accurately reflect the issue I found.

closed 2017-08-12 by mcnis003

KommandorKeen, 2017-08-20:
Thanks mcnis003 for closing this out, rather than fixing the issue and remaining silent here. It is only by the kind of collaborative behaviour exhibited by yourself that these projects prosper.