OpenDroneMap/ODM #652: Orthophotos mismatch at bordering area

hy9be, 2017-08-17:
Hi! I am trying to follow the "splits my dataset into 80 manageable chunks" strategy discussed in smathermather's [article](https://smathermather.com/2017/03/08/scaling-opendronemap-necessary-and-fun-next-steps/).

[image]

I tried to group images into different tasks under the same project in WebODM. But the result I got has some mismatch at the borders of orthophotos from different chunks, even they actually use the same image files at those area (see the image below).

Any suggestions to improve? Thanks!

[image]

KommandorKeen, 2017-08-17:
Did you have overlap between the groups?
See attached sketch
Each group needs to be  trimmed at constrained rather than unconstrained borders.  
[image]

hy9be, 2017-08-17:
@KommandorKeen I do not have that many overlaps (only 5 instead of 15, as below). And all my images are taken vertically.

[image]

KommandorKeen, 2017-08-17:
OK, try overlapping the photos so that g1 and g2 share 2 rows of photos then trim orthomosaics so they join between the two shared lines. If that is more promising, but not perfect, make g1 and g2 share 3 rows of photos then trim the orthomosaics along the line of the middle overlapping rows.

hy9be, 2017-08-18:
Basically try to have more overlapping? Am I getting it right?

I do not quite understand how the two tile can mismatch at the border, as they share the same image files. Could you give me some ideas? Thx!

KommandorKeen, 2017-08-18:
Think of all the images stitched together as a big rubber sheet. The sheet is stretched out by all triangular relationships between cameras and identifiable points on the images. Nothing actually fits perfectly and the tension is uneven across the sheet. Each camera position is well defined by links to many other cameras and image points, Once you move outside of the area where tension on a point can come from opposite directions (looking down - has a camera on 2 of more opposing sides of it) then the position is less constrained and the rubber sheet can warp in unexpected ways. Each group of images have different tie point relations and therefore different tensions causing different distortions.  Hope this helps.

hy9be, 2017-08-18:
Thanks @KommandorKeen !

closed 2017-08-18 by hy9be

hy9be, 2017-08-18:
@KommandorKeen I read several papers today and I think I got the idea. One more question (maybe a pretty dummy one..), by saying "trim orthomosaics so they join between the two shared lines", do you mean I should manually edit the orthomosaics?

KommandorKeen, 2017-08-20:
Yup, if you load the orthomosaics into qgis and use the geotag and import photos plugin you can see where the photos are and create a polygon (or series of polygons) to trim to.
using the raster - extraction - clipper function

smathermather, 2017-08-20:
@hy9be -- you might try these scripts to split and re-merge your dataset: https://github.com/paulinus/OpenDroneMap/tree/feature-split-dataset/scripts/metadataset. They're not in master yet, but they work quite well. See also: https://www.dropbox.com/s/7c3b3k4us2t8e3i/demo_zanzibar.mp4?dl=0

hy9be, 2017-08-21:
@smathermather I tried the ``./run_all.sh`` but runs into some runtime errors. Should these scripts totally replace the reconstruction steps in odm_app?