OpenDroneMap/ODM #653: Point cloud generation noisy

KommandorKeen, 2017-08-17:
[image]

ODM on left colmap on the right.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/point-cloud-generation-noisy/366

smathermather, 2018-11-22:
Discussion to date has been around improving the point cloud algorithm, filtering the points, and filtering on resultant products.

Noise reduction is needed, though it's not entirely clear where. I'm bringing this issue back as I explore this over the next few weeks. Pulling this from the [community post by Adam Steer](http://community.opendronemap.org/t/point-cloud-generation-noisy/366/2).

[image]

Noise reduction on the DSM generated from these data has limited value without noise reduction further back in the toolchain:
[image]
[image]

smathermather, 2018-11-22:
Utilities for noise reduction in points and rasters: https://personalpages.manchester.ac.uk/staff/neil.mitchell/mdenoise/ (via @pierotofy).

smathermather, 2018-11-22:
Next week, looking to test the following:
* Test changes to SMVS gamma
* Test use of OpenMVE for point cloud
* Test use of COLMAP for point cloud
* Test mdenoise on point clouds individually before merging if possible.
* Test mdenoise on point clouds once merged

pierotofy, 2021-04-05:
Closing, we've made lots of improvements to point clouds and DEMs since this was first opened. :+1: 

I'm willing to bet we currently outperform / or at least match COLMAP.

closed 2021-04-05 by pierotofy

smathermather, 2021-04-05:
Agreed: this issue has been thoroughly buried. :smile: