OpenDroneMap/ODM #656: What is the status of Classified Point Clouds

AsimAziz, 2017-08-24:
Hi, 

Sir I am successfully Classify Las Files as Ground Vs Non Ground But i want classification with further classes like **buildings,Power Lines, vegetation, vehicles, and water**.As required unit case is , I have to classify this in given point cloud.

Can You Please Inform Status of Classified Point Cloud   And  Please Also Explain Either Classification Means elements **Classification  [buildings,Power Lines, vegetation, vehicles, and water]** OR  JUST
**Ground Vs NON GROUND Classification**.

dakotabenjamin, 2017-08-24:
Currently we only classify ground vs non-ground. I would greatly welcome contributions to this end and can answer questions as to where in the code the classification process is.

pierotofy, 2017-09-01:
Classification happens in the lidar2dems code (via PDAL). Only ground vs non-ground. I don't think there are robust methods to classify automatically other types of objects. It's an area of active research.

KommandorKeen, 2018-02-27:
The karttapullautin code  is the best 'general' classification code I have come across. But most of the time I will use Canupo in Cloud Compare as this allows me to define the point classes easily. 
ANOTHER alternative is to grid the data at a suitable resolution and calculate/collect:
Average Height,Height SD, Height spread, number of points, Red, Green, Blue, (sum of RGB), normalised R,G and B, for each cell and then run a clustering algorithm.

smathermather, 2018-03-16:
@KommandorKeen -- does karttapullautin do classification of use an existing classified cloud?

I've been facinated by Canupo too -- it seems promising for shape-based classification. I know folks have also done some advanced classifications in PDAL using the variety of filters there.

pierotofy, 2018-03-17:
Keep an eye on CGAL also... their latest beta features:

```
Classification (new package)
This package offers an algorithm that classifies a data set into a user-defined set of labels (such as ground, vegetation, buildings, etc.). A flexible API is provided so that users can classify any type of data, compute their own local features on the input data set, and define their own labels.
```

https://www.cgal.org/2018/02/27/cgal412-beta1/

No docs yet, so maybe will wait for a stable release before looking into it.

smathermather, 2018-03-17:
Woah!

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/what-is-the-status-of-classified-point-clouds/363

closed 2018-03-28 by dakotabenjamin

Sajjadmanal, 2019-08-06:
@AsimAziz Hi, just curious to know if you were able to classify buildings,Power Lines, vegetation, vehicles, and water too?