OpenDroneMap/ODM #658: odm_data_bellus dataset processing using docker

mrgloom, 2017-08-24:
I'm trying to process odm_data_bellus dataset
https://github.com/OpenDroneMap/odm_data_bellus

I was running processing like:
`docker run --cpus 8 -it --rm -v  ${BASE_PATH}/opendronemap_benchmark/data/images:/code/images -v ${BASE_PATH}/opendronemap_benchmark/data/odm_orthophoto:/code/odm_orthophoto -v ${BASE_PATH}/opendronemap_benchmark/data/odm_georeferencing:/code/odm_georeferencing opendronemap/opendronemap --opensfm-processes 8`

As I understand odm_data_bellus dataset have gps data directly in exif metadata of image.

But processing ended with some errors and I don't see any intermediated results in `images`,`odm_orthophoto`,`odm_georeferencing` folders so where it's located?
~~~
Building objmodel:
	Saving model... done.
Whole texturing procedure took: 111.129s
[94m[INFO]    Running ODM Texturing Cell - Finished[0m
[94m[INFO]    Running ODM Georeferencing Cell[0m
[92m[DEBUG]   None[0m
[93m[WARNING] No coordinates file. Generating coordinates file: /code/odm_georeferencing/coords.txt[0m
[93m[WARNING] Could not generate coordinates file. Ignore if there is a GCP file. Error: Error in OdmExtractUtm:
Failed to open /code/odm_georeferencing/coords.txt for writing.
For more detailed information, see log file.
[0m
[93m[WARNING] Georeferencing failed. Make sure your photos have geotags in the EXIF or you have provided a GCP file. [0m
[93m[WARNING] Found a valid georeferenced model in: /code/odm_georeferencing/odm_georeferenced_model.ply[0m
[94m[INFO]    Running ODM Georeferencing Cell - Finished[0m
[94m[INFO]    Running ODM DEM Cell[0m
[92m[DEBUG]   running l2d_classify --help > /dev/null[0m
[94m[INFO]    Create DSM: False[0m
[94m[INFO]    Create DTM: False[0m
[94m[INFO]    DEM input file /code/odm_georeferencing/odm_georeferenced_model.las found: False[0m
[93m[WARNING] DEM will not be generated[0m
[94m[INFO]    Running ODM DEM Cell - Finished[0m
[94m[INFO]    Running ODM Orthophoto Cell[0m
[92m[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /code/odm_texturing/odm_textured_model.obj -logFile /code/odm_orthophoto/odm_orthophoto_log.txt -outputFile /code/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCornerFile /code/odm_orthophoto/odm_orthophoto_corners.txt[0m
Error in OdmOrthoPhoto:
Argument '-logFile' has a bad value.
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_orthophoto.py", line 74, in process
    '-outputCornerFile {corners}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
~~~

[timeline_odm_data_bellus.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/1248414/timeline_odm_data_bellus.txt)

dakotabenjamin, 2017-08-24:
That's odd, bellus is usually a very good dataset. Have you looked through the images manually for any corrupted files?

mrgloom, 2017-08-24:
Looks like something wrong with parsing arguments and not with images itself:

```
[92m[DEBUG]   running /code/build/bin/odm_orthophoto -inputFile /code/odm_texturing/odm_textured_model.obj -logFile /code/odm_orthophoto/odm_orthophoto_log.txt -outputFile /code/odm_orthophoto/odm_orthophoto.png -resolution 20.0  -outputCornerFile /code/odm_orthophoto/odm_orthophoto_corners.txt[0m
Error in OdmOrthoPhoto:
Argument '-logFile' has a bad value.
Traceback (most recent call last):
  File "/code/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_orthophoto.py", line 74, in process
    '-outputCornerFile {corners}'.format(**kwargs))
  File "/code/opendm/system.py", line 34, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

https://github.com/OpenDroneMap/OpenDroneMap/blob/5d77726315f3c1393c9e770d2bf8cdef8e54a702/scripts/odm_orthophoto.py#L72

Looks like similair to https://github.com/OpenDroneMap/OpenDroneMap/issues/64 but I don't pass this parameter by myself.

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/odm-data-bellus-dataset-processing-using-docker/383

closed 2018-03-28 by dakotabenjamin