OpenDroneMap/ODM #659: SIGSEGV fault in odm_meshing

dakotabenjamin, 2017-08-28:
So i'm trying to update ODM so it will actually work with 16.10+ starting with #644 . 

gdb debug: 
```
Starting program: /home/ubuntu/opendronemap/build/bin/odm_meshing -inputFile /vagrant_data/Toledo/opensfm/depthmaps/merged.ply -outputFile /vagrant_data/Toledo/odm_meshing/odm_mesh.ply -logFile /vagrant_data/Toledo/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9 -verbose
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Successfully loaded 2876685 points with corresponding normals from file.
Octree depth used for reconstruction is: 9
Estimated initial vertex count: 262144

Reconstruction complete:
Vertex count: 273184
Triangle count: 546103

Decimating mesh, removing 63.3946 percent of vertices.
Decimation complete.
Decimated vertex count: 100073
Decimated triangle count: 199901

Breakpoint 1, OdmMeshing::writePlyFile (this=0x7fffffffe030) at /home/ubuntu/opendronemap/modules/odm_meshing/src/OdmMeshing.cpp:351
351     {
(gdb) next
352         log_ << "Saving mesh to file.\n";
(gdb)
Saving mesh to file.
353         if (pcl::io::savePLYFile(outputFile_.c_str(), *mesh_.get())  == -1) {
(gdb)
359                  << outputFile_ << "\n";
(gdb)
Successfully wrote mesh to:
/vagrant_data/Toledo/odm_meshing/odm_mesh.ply
361     }
(gdb)
OdmMeshing::run (this=0x7fffffffe030, argc=16, argv=0x7fffffffe388) at /home/ubuntu/opendronemap/modules/odm_meshing/src/OdmMeshing.cpp:79
79          log_.printToFile(logFilePath_);
(gdb)
80          return EXIT_SUCCESS;
(gdb)
81      }
(gdb)
main (argc=16, argv=0x7fffffffe388) at /home/ubuntu/opendronemap/modules/odm_meshing/src/main.cpp:17
17          OdmMeshing meshCreator;
(gdb)

Program received signal SIGSEGV, Segmentation fault.
__GI___libc_free (mem=0x91) at malloc.c:2965
2965    malloc.c: No such file or directory.
(gdb) bt
#0  __GI___libc_free (mem=0x91) at malloc.c:2965
#1  0x00005555555a7d7f in boost::checked_delete<pcl::Poisson<pcl::PointNormal> > (x=0x55555580fe70) at /usr/include/boost/core/checked_delete.hpp:34
#2  0x00005555555ae244 in boost::detail::sp_counted_impl_p<pcl::Poisson<pcl::PointNormal> >::dispose (this=0x5555557fc3e0) at /usr/include/boost/smart_ptr/detail/sp_counted_impl.hpp:78
#3  0x000055555559b0a0 in boost::detail::sp_counted_base::release (this=0x5555557fc3e0) at /usr/include/boost/smart_ptr/detail/sp_counted_base_gcc_x86.hpp:146
#4  0x000055555559b133 in boost::detail::shared_count::~shared_count (this=0x7fffffffe1c8, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:473
#5  0x00005555555a20ee in boost::shared_ptr<pcl::Poisson<pcl::PointNormal> >::~shared_ptr (this=0x7fffffffe1c0, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/shared_ptr.hpp:336
#6  0x000055555559ce76 in OdmMeshing::~OdmMeshing (this=0x7fffffffe030, __in_chrg=<optimized out>) at /home/ubuntu/opendronemap/modules/odm_meshing/src/OdmMeshing.cpp:26
#7  0x000055555559a8ac in main (argc=16, argv=0x7fffffffe388) at /home/ubuntu/opendronemap/modules/odm_meshing/src/main.cpp:17
(gdb) Quit
(gdb)

```

To recreate, in a 16.10 clean build, clone ODM and pull #644. then apply the following patch: https://gist.github.com/dakotabenjamin/8f022ef281bfac869a45ab6b82d848f9

I think there's something wrong deleting things from memory but I'm not well-versed enough in how memory management works in c++ to figure this out.

fredlllll, 2017-09-01:
it seems the error is happening in the destruction of the smart pointer to the pcl::Poisson object.

i saw that this is only used in OdmMeshing::createMesh. you could try to move the variable there and with that get around the smartpointer. after all there is no need for the object to be a member variable/smart pointer. also dynamic memory allocation isnt strictly necessary here

pierotofy, 2020-09-11:
This can be closed as odm_meshing is no longer used. :+1:

closed 2020-09-11 by pierotofy