OpenDroneMap/ODM #660: Stuck at "Merging depthmaps"

hy9be, 2017-08-30:
I tried to run a demo dataset of senseFly with ODM:

[image]

This is a big dataset with 180+ images. I tried on a very powerful aws machine with 32GB memory. But the docker stuck after prints the line ``Merging depthmaps``, for about 3 hours. I just stopped the docker as the CPU usage is very close to 0%.

I want to find out the issue behind but not sure where to start. Any suggestions? Thanks!

PS: you could download the files from [here](https://senseflycom.s3.amazonaws.com/datasets/ebee-rtk/images/download/img.7z).

mcnis003, 2017-08-30:
How much memory did you give Docker? This is a Docker setting. If you are at 0% CPU then it will not finish. Restart Docker with an increased RAM allotment and restart your process. Not sure how big your data set is but try giving it 16GB to start.

hy9be, 2017-08-30:
@mcnis003 I built and run ODM natively on that aws machine. Did not use docker. I assume the memory should be enough.

And I just tried it on my own desktop, with 12GB memory assigned to docker. Still stucks at "Merging depthmaps".

mcnis003, 2017-08-30:
@hy9be Here is the setups I use on my ~25mb datasets:

2016 12in MacBook - 2 CPUs 6GB RAM ~2 hour runtime
Google Cloud - 8 CPUs 32GB RAM ~40 minute runtime

If the requirements scale linearly (which I am not sure of), for a 800mb dataset I believe you would need:

2016 12in MacBook - 2 CPUs 192GB RAM ~64 hour runtime
Google Cloud - 8 CPUs 1024GB RAM ~21 hour runtime

Just to double check, I went to DroneDeploy and I can see that their documentation has an example where the dataset is 120mb and they report that it would take ~6 hours for them to process this dataset. I am not sure what setup they use to process their datasets, but I am pretty sure your 800mb+ dataset is a monster.

https://support.dronedeploy.com/v1.0/docs/how-to-process-datasets#section-data-requirements

Hopefully someone who has actually ran a larger job like this could comment. Maybe there are other variables that you could set to run your dataset more efficiently. I am not experienced with running a larger dataset like this, I can only comment on what I believe the requirements would be based on my experience on my machine and with running the same datasets on gcloud.

Whites your objective with running this dataset? If your goal is to prove that ODM working on your machine I would either:
1. subset this data set to a small number of photos (like 9 that form a square)
2. choose a different dataset that is much smaller.

hy9be, 2017-08-31:
@mcnis003 I doubt this is a computation resource issue, for two reasons:

1. The memory usage of the docker on my macbook is about 9GB. There are still some unused memories.
2. I lowered my docker memory cap to 2GB and tried the same dataset, odm process exits with error, rather than hanging at some step.

And as you are not sure of, your linear extrapolation on system resource does not make sense. Even the requirements scale linearly, you will need at least two data points for that kind of "linear regression". For example, 25mb, 100mb, and then 800mb. 

Also the dataset size is not the only variable that impact the computation resources needed.

dakotabenjamin, 2017-08-31:
#644 should fix the merging depthmaps issue. Can you test?

dakotabenjamin, 2017-09-01:
MErging issue has been fixed, pull the latest changes from master. Please reopen if the changes don't fix your issue.

closed 2017-09-01 by dakotabenjamin