OpenDroneMap/ODM #662: Exposing depth map options

dakotabenjamin, 2017-08-31:
Here are the OpenSfM depth map options that can be exposed to ODM:
```
depthmap_method: PATCH_MATCH          # Raw depthmap computationg algorithm (PATCH_MATCH, BRUTE_FORCE, PATCH_MATCH_SAMPLE)
depthmap_resolution: 640              # Resolution of the depth maps
depthmap_num_neighbors: 10            # Number of neighboring views
depthmap_num_matching_views: 2        # Number of neighboring views used for each depthmaps
depthmap_patchmatch_iterations: 3     # Number of PatchMatch iterations to run
depthmap_min_patch_sd: 5.0            # Patches with lower stardard deviation are ignored
depthmap_min_correlation_score: 0.7   # Minimum correlation score to accept a depth value
depthmap_same_depth_threshold: 0.005  # Threshold to measure depth closeness
depthmap_min_consistent_views: 3      # Min number of views that should reconstruct a point for it to be valid
depthmap_save_debug_files: no         # Save debug files with partial reconstruction results
```

Testing needs to be done to see how these affect the process/quality of reconstruction.

pierotofy, 2018-02-24:
Ups, just realized this was already here. I opened a duplicate on #769

smathermather, 2018-03-14:
This will work so much better than me changing the configuration file every time... :)

smathermather, 2019-11-14:
Closing as OpenSfM isn't the primary depthmap engine now, and adding more flags for a non-primary depthmap engine seems excessive. Feel free to re-open or discuss.

closed 2019-11-14 by smathermather