OpenDroneMap/ODM #663: Method to get the suitable resolution of the orthophoto

xialang2012, 2017-09-01:
The current method which set the resolution of the orthophonto to 20 point/m is not very flexible. We can 
 calculate it from the input image. If an image has the GPS information, then the resolution can be calculated by the below equation.

Resolution = 100 / (H * SizeofPixel / focalLength)

Here, H is the altitude of the flight (m), SizeofPixel is the size of the pixel for the CMOS (micro-meter), focalLength is the focal length (mm). The unit of Resolution is points/m. 

@dakotabenjamin

KommandorKeen, 2017-09-01:
There was me thinking it was pretty flexible. Sometimes i don't want a full resolution image. 50 hectares at 2500 pixels   /m2 is just too many.

dakotabenjamin, 2017-09-01:
Perhaps we can find a solution that works for both. Use @xialang2012 suggestion for the default but give the ability to scale down as needed.

pierotofy, 2019-05-23:
@xialang2012 we've added better orthophoto units (cm / pixels) and constraints that use GSD estimates to better calculate a suitable resolution for the orthophoto. Please update and let us know if it fits your use case? If not, please re-open the issue :hugs:

closed 2019-05-23 by pierotofy